joan2937/pigpio #85: Bug report: 'int level' argument when calling callback function from ISR

CVino, 2016-10-01:
I'm having an issue when setting a callback function with "gpioSetISRFuncEx(unsigned user_gpio, unsigned edge, int timeout, gpioISRFuncEx_t f, void *userdata)" .  e.g.:

gpioSetISRFuncEx(_echo, EITHER_EDGE, 0, WrapperRecordPulse,this);

My "WrapperRecordPulse" function is called back with the next arguments: 

WrapperRecordPulse(int gpio, int level, uint32_t tick,void* Pointer)

where "int level" is the rising (RISING_EDGE) or falling (FALLING_EDGE) edge returned by the interrupt call. The problem that I found in my code is that the "int level" argument is the opposite as it might be, that is, the Interrupt calls the callback function with "int level" = RISING_EDGE when there is a falling edge and viceversa.

joan2937, 2016-10-01:
Could you provide a small example program with build instructions which demonstrates the problem?

joan2937, 2016-10-01:
The following works correctly.
```C
#include <stdio.h>
#include <stdlib.h>

#include <pigpio.h>

/*
gcc -Wall -pthread -o ISR-test ISR-test.c -lpigpio

sudo ./ISR-test 0 
sudo ./ISR-test 1 
sudo ./ISR-test 2 

To test do the following while ISR-test is running.

for ((i=0; i<10; i++)); do pigs w 4 1 w 4 0 mils 1; done
*/

#define GPIO 4

void isr_ex(int gpio, int level, uint32_t tick, void *userdata)
{
   int now = gpioRead(gpio);
   printf("g=%d l=%d(%d) t=%u ud=%d\n",
      gpio, level, now, tick, *(int*)userdata);
}

int main(int argc, char *argv[])
{
   int userdata = 543, edge = FALLING_EDGE;

   if (argc > 1)
   {
      edge = atoi(argv[1]);
      if ((edge < 0) || (edge >2)) edge = 0;
   }

   if (gpioInitialise() < 0) return -1;

   gpioSetISRFuncEx(GPIO, edge, 0, isr_ex, &userdata);

   time_sleep(60);

   gpioSetISRFuncEx(GPIO, edge, 0, NULL, &userdata);

   gpioTerminate();

   return 0;
}
```
```
$ sudo ./ISR-test 0
g=4 l=1(1) t=3116182478 ud=543
g=4 l=1(1) t=3116204318 ud=543
g=4 l=1(1) t=3116231069 ud=543
g=4 l=1(1) t=3116252457 ud=543
g=4 l=1(1) t=3116276549 ud=543
g=4 l=1(1) t=3116298853 ud=543
g=4 l=1(1) t=3116323643 ud=543
g=4 l=1(1) t=3116352645 ud=543
g=4 l=1(1) t=3116375821 ud=543
^C2016-10-01 11:27:52 sigHandler: Unhandled signal 2, terminating

$ sudo ./ISR-test 1
g=4 l=0(0) t=3127520038 ud=543
g=4 l=0(0) t=3127544186 ud=543
g=4 l=0(0) t=3127566150 ud=543
g=4 l=0(0) t=3127588990 ud=543
g=4 l=0(0) t=3127611819 ud=543
g=4 l=0(0) t=3127636381 ud=543
g=4 l=0(0) t=3127670654 ud=543
g=4 l=0(0) t=3127700775 ud=543
g=4 l=0(0) t=3127731240 ud=543
g=4 l=0(0) t=3127762465 ud=543
^C2016-10-01 11:28:03 sigHandler: Unhandled signal 2, terminating

$ sudo ./ISR-test 2
g=4 l=0(0) t=3136671625 ud=543
g=4 l=1(1) t=3136692870 ud=543
g=4 l=0(0) t=3136697818 ud=543
g=4 l=1(1) t=3136716097 ud=543
g=4 l=0(0) t=3136720076 ud=543
g=4 l=1(1) t=3136741758 ud=543
g=4 l=0(0) t=3136746002 ud=543
g=4 l=1(1) t=3136764796 ud=543
g=4 l=0(0) t=3136768969 ud=543
g=4 l=1(1) t=3136786809 ud=543
g=4 l=0(0) t=3136791105 ud=543
g=4 l=1(1) t=3136812816 ud=543
g=4 l=0(0) t=3136817420 ud=543
g=4 l=1(1) t=3136836266 ud=543
g=4 l=0(0) t=3136840621 ud=543
g=4 l=1(1) t=3136861505 ud=543
g=4 l=0(0) t=3136865867 ud=543
g=4 l=1(1) t=3136884129 ud=543
g=4 l=0(0) t=3136888346 ud=543
^C2016-10-01 11:28:52 sigHandler: Unhandled signal 2, terminating

```

CVino, 2016-10-01:
Sure, here is one I just made as a test:

/* RPI CONNECTIONS: 
 * Please connect pin 38 (GPIO 20)  and pin 40 (GPIO 21) together
 * 
 * BUILD COMMAND:
 * g++ -std=c++14  -Wall -pthread -o "%e" "%f" -lpigpio -lrt 
 * 
 * */

#define pin1 20
#define pin2 21

#include <pigpio.h>
#include <iostream>

using namespace std;

void RecordPulse(int, int, uint32_t,void*);

int main(void)
{
	if (gpioInitialise() <0)
	{
		std::cout <<"Initialisation error of the GPIO \n Closing program..."<< std::endl;
		return (-1);
	}

	int* MyData;  //Data pointer for the callback function (non initialized and not used)
	
	gpioSetMode(pin1,PI_OUTPUT);  //Set pin1 as output
	gpioWrite(pin1, 0);			//Set pin1 output to 0
	gpioSetMode(pin2,PI_INPUT); //Set pin2 as input
	gpioSetISRFuncEx(pin2, EITHER_EDGE, 0, RecordPulse,(int*) MyData); //Register Callback function (interrupt with both edges)
	
	gpioSleep(PI_TIME_RELATIVE, 1, 0);  //Delay 1 second
	gpioWrite(pin1, 1);			        //Set pin1 output to 1 (Rising Edge)
	gpioSleep(PI_TIME_RELATIVE, 1, 0);  //Delay 1 second
	gpioWrite(pin1, 0);			        //Set pin1 output to 0 (Falling Edge)
}

void RecordPulse(int gpio, int level, uint32_t tick,void* Data)
{
	int* MyData = (int*)Data;  //Passed Data. Not used!
	
	if (level == RISING_EDGE)
	cout << "(1)This is a rising one" << endl; //Should appear in first place
	if(level == FALLING_EDGE)
	cout << "(2)This is a falling one" << endl;//Should appear in second place
}

CVino, 2016-10-01:
Here I just made a really easy example to prove it:

`		/* RPI CONNECTIONS: 
		 * Please connect pin 38 (GPIO 20)  and pin 40 (GPIO 21) together
		 * 
		 * BUILD COMMAND:
		 * g++ -std=c++14  -Wall -pthread -o "%e" "%f" -lpigpio -lrt 
		 * 
		 * */

		#define pin1 20
		#define pin2 21

		#include <pigpio.h>
		#include <iostream>

		using namespace std;

		void RecordPulse(int, int, uint32_t,void*);

		int main(void)
		{
			if (gpioInitialise() <0)
			{
				std::cout <<"Initialisation error of the GPIO \n Closing program..."<< std::endl;
				return (-1);
			}

			int* MyData;  //Data pointer for the callback function (non initialized and not used)
			
			gpioSetMode(pin1,PI_OUTPUT);  //Set pin1 as output
			gpioWrite(pin1, 0);			//Set pin1 output to 0
			gpioSetMode(pin2,PI_INPUT); //Set pin2 as input
			gpioSetISRFuncEx(pin2, EITHER_EDGE, 0, RecordPulse,(int*) MyData); //Register Callback function (interrupt with both edges)
			
			gpioSleep(PI_TIME_RELATIVE, 1, 0);  //Delay 1 second
			gpioWrite(pin1, 1);			        //Set pin1 output to 1 (Rising Edge)
			gpioSleep(PI_TIME_RELATIVE, 1, 0);  //Delay 1 second
			gpioWrite(pin1, 0);			        //Set pin1 output to 0 (Falling Edge)
		}

		void RecordPulse(int gpio, int level, uint32_t tick,void* Data)
		{
			int* MyData = (int*)Data;  //Passed Data. Not used!
			
			if (level == RISING_EDGE)
			cout << "(1)This is a rising one" << endl; //Should appear in first place
			if(level == FALLING_EDGE)
			cout << "(2)This is a falling one" << endl;//Should appear in second place
		}
`

CVino, 2016-10-01:
$ sudo ./Ex
    (2)This is a falling one
    (1)This is a rising one

joan2937, 2016-10-01:
You appear to be confusing the constants used to define whether to set up a rising edge, falling edge, or either edge interrupt with the level of the GPIO after the edge happens.

A falling edge means a transition from high to low (1 to 0).  A rising edge means a transition from low to high (0 to 1) and either edge means a level change (0 to 1 or 1 to 0).

CVino, 2016-10-01:
Thats clear to me. In my example, there is a first transition 0 to 1 (rising edge) on first place and then after the "gpioWrite(pin1,0)" there is a falling edge. In the stdout you can see that the order is reverse.

In your example, on the first run f.e. (./ISR-Test 0) you configure the interrupt to detect a RISING_EDGE (0) but afterwards you read the pin and it appears to be "0" so there is not such a rising edge.

CVino, 2016-10-01:
As per the definition of "edge" that is as follows:

edge: 0-2

The type of GPIO edge to generate an intrrupt. See gpioSetISRFunc, and gpioSetISRFuncEx. 

RISING_EDGE 0
FALLING_EDGE 1
EITHER_EDGE 2

CVino, 2016-10-01:
I think I got it now. "level" is the gpioWrite just after the interrupt right?

joan2937, 2016-10-01:
You are confusing the constants used to define the type of interrupt wanted with the level of the GPIO.

If you ask to be told about a RISING_EDGE you expect to be told when the GPIO becomes high.
If you ask to be told about a FALLING_EDGE you expect to be told when the GPIO becomes low.
If you ask to be told about EITHER_EDGE you expect to be told when the GPIO changes level.

CVino, 2016-10-01:
Yes I get that, the level is then the state of the gpio after the edge right?

joan2937, 2016-10-01:
"Yes I get that, the level is then the state of the gpio after the edge right?"

Usually, but.

> The tick is that read at the time the process was informed of the interrupt. This will be a variable number of microseconds after the interrupt occurred. Typically the latency will be of the order of 50 microseconds. The latency is not guaranteed and will vary with system load.
> 
> The level is that read at the time the process was informed of the interrupt, or PI_TIMEOUT if the optional interrupt timeout expired. It may not be the same as the expected edge as interrupts happening in rapid succession may be missed by the kernel (i.e. this mechanism can not be used to capture several interrupts only a few microseconds apart).

CVino, 2016-10-01:
Oh, good to know that! How many uSec are 'a few microseconds apart'? my
process can have a pulse of around 200 uSec and I must register both edges.
Would that be a problem?

Sorry for the missunderstanding and opening the issue.

I appreciate your time and good job with this library, really useful.

joan2937, 2016-10-01:
Anything less than 50µs or so if repeated regularly would be a problem for kernel interrupts.

Unless you need absolutely the minimum latency (or the pulse length is shorter than the sample period) you would be better off using alerts rather than interrupts.

Alerts will have a more accurate timestamp and can handle interrupts in rapid succession.

CVino, 2016-10-01:
Thanks for the hint!

closed 2016-10-01 by joan2937

stheine, 2016-10-01:
In the pigpio C API, there is gpioSetISRFunc() and gpioSetAlertFunc().
In the pigpiod C API, there is only callback(). is this ISR or Alert based? does this have the mentioned issues as gpioSetISRFunc()?

joan2937, 2016-10-01:
The pigpiod C API is used to control the GPIO via the running pigpio daemon over a socket interface.  I didn't see any point in exposing the Linux interrupt functionality as any latency gain would be swamped by the network message time.

stheine, 2016-10-01:
so you're saying, callback() is the equivalent to gpioSetAlertFunc(), right?

joan2937, 2016-10-01:
"so you're saying, callback() is the equivalent to gpioSetAlertFunc(), right?"

Yes, that is correct.