joan2937/pigpio #86: pigpio for Allwinner A20 ou H3

seblucas, 2016-10-03:
Hi,

I'm using pigpio for a long time with my raspberry pis, For various reason I also use Orange Pi (Allwinner H3) / Banana Pi (A20) and I'm really missing pigpio. pigpio is compiling perfectly on both but I never tried to use it fearing to damage my boards (as many GPIO are in use).

So I have two questions : 
 * What could I do to support other board ? is there a file in particular to modify ? Are you interested in it ?
 * I decided to write this issue after spending 2,5 hours to translate a python program reading a si7021 through i2c using pigpio (on an old Raspberry pi) to a simple python script using smbus and I had to keep only temperature / humidity readings, everything else was erroring. Am I taking a risk if I only use pigpio only for it's i2c primitives ? and spi ?

Thanks in advance.

joan2937, 2016-10-03:
The interesting stuff pigpio does requires DMA, in particular the ARM DMA hardware build into the Broadcom SoCs used by the Pi.

The DMA support is used to provide the high rates of multiple GPIO sampling, waves, and software serial output.

Ordinary SPI, I2C, and serial were very much an afterthought and were principally added to make that functionality if the Pi was being controlled over a network.

Changing SPI and serial to use a a different platform are pretty much name change edits.

I2C is similar, most of the functions are direct implementations of standard Linux SMBus commands.  Again just a name change edit.  Two exceptions are i2c_device_read and i2c_device_write which talk directly to the Linux I2C device (normally /dev/i2c-1).  They can be converted to standard Python if the ioctl module is used.

Unless a clone has a Broadcom SoC I see little possibility of a port.

seblucas, 2016-10-03:
Thanks a lot for your answer. I guess I'll continue to use C with my Allwinner stuff, Fast Python will be a Raspberry thing ;)

Thanks again

closed 2016-10-03 by seblucas