joan2937/pigpio #87: gpioInitialise cause I2S DAC sound card does not work properly

howardqiao, 2016-10-04:
gpioInitialise cause I2S interface DAC sound card does not work properly, I do not know you have not tried, when MPD playing music, run gpioInitialise, will cause the player to stop, when the replay, the sound is not normal.

joan2937, 2016-10-04:
Yes, it will do.

The I2S peripheral uses the PCM hardware on the Pi.  By default pigpio also uses the PCM hardware to time DMA transfers.

You can tell pigpio to use the PWM hardware instead of the PCM hardware.  You do that by calling [gpioCfgClock](http://abyz.co.uk/rpi/pigpio/cif.html#gpioCfgClock) **before** calling gpioInitialise.
.

E.g.
```C
gpioCfgClock(5, 0, 0);
if (gpioInitialise() < 0) exit(-1);
...
gpioTerminate();
```
Of course this means you can't use the PWM hardware for ordinary audio.

N.B.  if you use the wave functions that forces the use of the other PWM/PCM hardware. So you will still lose PCM if you use waves from pigpio.  If you don't know what waves are you probably aren't using them.

closed 2016-10-05 by joan2937

NBroomfield, 2021-04-03:
I'm having the same issue, I added gpioCfgClock(5, 0, 0) which seem to fix the issue but now my comms via /dev/serial0 are messed up, any suggestions? I'm not actually using pigpio for anything complex just reading an writing to some pins, so I don't need complex PWM or interrupts on pins.

Thanks