joan2937/pigpio #88: calling `pi.stop` does not stop hardware PWM

yosemitebandit, 2016-10-05:
I'm using a Raspberry Pi 3.  The hardware PWM functionality is fantastic, just wondering if there is a proper way to cleanup / call a "destructor."

Steps to reproduce:

```
$ sudo pigpiod
$ python
>>> import pigpiod
>>> pi = pigpiod.pi()
>>> pi.hardware_PWM(13, 200, 300000)  # 30% duty cycle at 200Hz on pin 13
>>> pi.stop()  # signal continues being sent on pin 13 until device is shutdown
```

joan2937, 2016-10-06:
If a script claims a resource or triggers an action then that resource will not be freed and that action will not be stopped unless the script explicitly releases the resource or explicitly stops the action.  Exiting a script has no action other than terminating the socket connection to the pigpio daemon

Generally the pigpio library makes no assumptions about the intentions of a user.

About the only exception I can think of is that if you write to a GPIO then pigpio assumes you want the GPIO to be an output so will switch the GPIO to be in mode OUTPUT.

You need to have your script do any necessary tidying up.

If the script terminates without doing the tidy up you can use pigs to do the tidy up.  E.g. the pigs [HP](http://abyz.co.uk/rpi/pigpio/pigs.html#HP) command can be used to stop hardware PWM.

yosemitebandit, 2016-10-06:
Ok, I will just make sure to tidy up as best as possible before exiting, thanks!

closed 2016-10-06 by yosemitebandit

nh-yoz, 2023-09-01:
Put the mode of the used pwm pin to "READ". Using pigs in a bash script:
```
#!/bin/bash
GPIO=18

trap "DoExit" EXIT

DoExit() {
    pigs M $GPIO R
}
```