joan2937/pigpio #89: serRead can write past the end of the passed buffer.

paeryn, 2016-10-09:
At the end of **serRead()** you forcibly do a
```
    buf[r] = 0;
    return r;
```
when the read succeeds which will write past the end of the buffer if the buffer is only big enough to hold the requested number of bytes and there are enough bytes to fulfil the read. The data being read isn't necessarily a string that needs null-terminating. I think it would be better to either only null-terminate if fewer than the requested number of bytes were read or state in the documentation that the buffer needs to be at least 1 byte longer than the requested read size.

joan2937, 2016-10-09:
You are quite correct.

As you say there is no reason to terminate the string as it's potentially binary anyhow.

It may be a testing hangover where it would have been convenient to null terminate test strings.

Because of potential breakage I'm making the following change to [serRead()](http://abyz.co.uk/rpi/pigpio/cif.html#serRead).
```C
      if (r < count) buf[r] = 0;
      return r;
```
The effect is to terminate as a string when possible and do nothing if the receive buffer is full.

This will be in V57.  I have no timescale for pushing out V57.

joan2937, 2016-10-30:
Should be okay now in V57.

closed 2016-11-01 by joan2937