joan2937/pigpio #90: Second call of gpioSetTimerFunc doesn't work in some cases

absalom-muc, 2016-10-28:
Hi joan,

with the following test program I have a strange effect with pigpio v56:

```
#include <stdio.h>
#include <pigpio.h>

void Timer0()
{
	printf("Timer0 triggered\n");
}

void Timer1()
{
	printf("Stop timer0\n");
	gpioSetTimerFunc(0, 500, NULL); // deactivate timer0
	gpioSetTimerFunc(1, 2000, NULL); // deactivate timer1
}

int main() {
	if(gpioInitialise() == PI_INIT_FAILED)
	{
		printf("gpioInitialise failed!\n");
		return -1;
	}
	else
		printf("gpioInitialise o.k.\n");
	
	printf("1st trial\n");
	gpioSetTimerFunc(0, 500, Timer0);
	gpioSetTimerFunc(1, 2000, Timer1);
	gpioDelay(5*1e6);
	printf("2nd trial\n");
	gpioSetTimerFunc(0, 500, Timer0);
	gpioSetTimerFunc(1, 2000, Timer1);
	while(1);
	
	gpioTerminate();
	return 0;
}
```
output is
```
gpioInitialise o.k.
1st trial
Timer0 triggered
Timer0 triggered
Timer0 triggered
Timer0 triggered
Stop timer0
2nd trial
Timer0 triggered
Timer0 triggered
Timer0 triggered
Timer0 triggered
Timer0 triggered
Timer0 triggered
...
```
As you can see the 1st trial works without problems, but the 2nd call of gpioSetTimerFunc(1, 2000, Timer1); seems to have no effect (return value is 0). Timer1 is not called to stop Timer0.

Could you have a look on it?
Thank you.
absalom

joan2937, 2016-10-28:
It's definitely behaviour I would prefer not to happen so from that point of view it is a bug.

It's probably related to killing the timer within itself.  When I get time I'll look in more detail.  If it is easy to allow this action I will do so, if it isn't I'll probably just add a note to say don't kill a timer within a timer.

```C
#include <stdio.h>
#include <pigpio.h>

void Timer0()
{
    printf("Timer0 triggered\n");
}

void Timer1()
{
    printf("Timer1 triggered\n");
    printf("Stop timer0\n");
    gpioSetTimerFunc(0, 500, NULL); // deactivate timer0
    //gpioSetTimerFunc(1, 2000, NULL); // deactivate timer1
}

int main()
{
    if(gpioInitialise() == PI_INIT_FAILED)
    {
        printf("gpioInitialise failed!\n");
        return -1;
    }
    else
        printf("gpioInitialise o.k.\n");

    printf("1st trial\n");
    gpioSetTimerFunc(0, 500, Timer0);
    gpioSetTimerFunc(1, 2000, Timer1);

    time_sleep(10);

    printf("2nd trial\n");
    gpioSetTimerFunc(0, 500, Timer0);
    gpioSetTimerFunc(1, 2000, Timer1);

    time_sleep(10);

    gpioTerminate();

    return 0;
}
```

```
$ sudo ./issue90
gpioInitialise o.k.
1st trial
Timer0 triggered
Timer0 triggered
Timer0 triggered
Timer0 triggered
Timer1 triggered
Stop timer0
Timer1 triggered
Stop timer0
Timer1 triggered
Stop timer0
Timer1 triggered
Stop timer0
Timer1 triggered
Stop timer0
2nd trial
Timer0 triggered
Timer0 triggered
Timer0 triggered
Timer1 triggered
Stop timer0
Timer1 triggered
Stop timer0
Timer1 triggered
Stop timer0
Timer1 triggered
Stop timer0
Timer1 triggered
Stop timer0
$ 
```

joan2937, 2016-10-28:
Okay, I have made a change which seems to work.  The following code/output shows the new behaviour.

The change will be in V57 which I'll push out in a day or two.

```C
#include <stdio.h>
#include <pigpio.h>

void Timer0()
{
    printf("Timer0 triggered\n");
}

void Timer1()
{
    printf("Timer1 triggered\n");
    printf("Stop timer0\n");
    gpioSetTimerFunc(0, 500, NULL); // deactivate timer0
    printf("Stop timer1\n");
    gpioSetTimerFunc(1, 2000, NULL); // deactivate timer1
}

int main()
{
   int i;

   if(gpioInitialise() == PI_INIT_FAILED)
   {
      printf("gpioInitialise failed!\n");
      return -1;
   }
   else
      printf("gpioInitialise o.k.\n");

   time_sleep(4);

   for (i=1; i<=10; i++)
   {
      printf("trial %d\n", i);
      gpioSetTimerFunc(0, 500, Timer0);
      gpioSetTimerFunc(1, 2000, Timer1);

      time_sleep(4);
   }

   gpioTerminate();

   return 0;
}
```

```
$ sudo ./issue90  
gpioInitialise o.k.
trial 1
Timer0 triggered
Timer0 triggered
Timer0 triggered
Timer0 triggered
Timer1 triggered
Stop timer0
Stop timer1
trial 2
Timer0 triggered
Timer0 triggered
Timer0 triggered
Timer0 triggered
Timer1 triggered
Stop timer0
Stop timer1
trial 3
Timer0 triggered
Timer0 triggered
Timer0 triggered
Timer0 triggered
Timer1 triggered
Stop timer0
Stop timer1
trial 4
Timer0 triggered
Timer0 triggered
Timer0 triggered
Timer0 triggered
Timer1 triggered
Stop timer0
Stop timer1
trial 5
Timer0 triggered
Timer0 triggered
Timer0 triggered
Timer0 triggered
Timer1 triggered
Stop timer0
Stop timer1
trial 6
Timer0 triggered
Timer0 triggered
Timer0 triggered
Timer0 triggered
Timer1 triggered
Stop timer0
Stop timer1
trial 7
Timer0 triggered
Timer0 triggered
Timer0 triggered
Timer0 triggered
Timer1 triggered
Stop timer0
Stop timer1
trial 8
Timer0 triggered
Timer0 triggered
Timer0 triggered
Timer0 triggered
Timer1 triggered
Stop timer0
Stop timer1
trial 9
Timer0 triggered
Timer0 triggered
Timer0 triggered
Timer0 triggered
Timer1 triggered
Stop timer0
Stop timer1
trial 10
Timer0 triggered
Timer0 triggered
Timer0 triggered
Timer0 triggered
Timer1 triggered
Stop timer0
Stop timer1
$ 
```

absalom-muc, 2016-10-28:
Great, thank you!

joan2937, 2016-10-30:
Should be okay now in V57.

absalom-muc, 2016-10-31:
Thank you joan

closed 2016-11-01 by joan2937