joan2937/pigpio #92: Start without forking

jdiamond, 2016-10-30:
Hi, is it possible to start pigpiod without forking a child process? That makes it difficult to run inside a Docker container because the parent process exits right away so the container stops.

joan2937, 2016-10-30:
I don't know.

This is not an area I know anything about.  I just copied what appeared to be standard daemon launching code at the time I wrote the daemon.

Does anyone know what current best practice is?

jw3, 2016-10-30:
@jdiamond I wonder if https://github.com/Yelp/dumb-init (or something like it) would help with pigpio as-is?

MrYsLab, 2016-10-30:
+1 on dumb-init

jdiamond, 2016-10-30:
dumb-init would be great, but it won't work with pigpiod because it forks. As soon as the parent pigpiod process exits, dumb-init exits. The child process is left running.

I have pigpiod running in a container using systemd. It has a `Type=forking` option for dealing with daemons like this, but having systemd running in a container feels so "heavy". I'd much rather use something lightweight like dumb-init. Especially on a Pi.

I think this is a great list of best practices for writing daemons:

https://web.archive.org/web/20160304120912/http://homepage.ntlworld.com/jonathan.deboynepollard/FGA/unix-daemon-design-mistakes-to-avoid.html

It's not specific to Docker, but it explains why forking is problematic.

If I'm feeling adventurous, I may try to add an option for this.

joan2937, 2016-10-30:
Frankly the unix daemon design mistakes to avoid article went over my head.

pigpiod my have many socket connections in use.  I'm not sure how that relates to using a helper process and just using stdin and stdout.  That sounds like only one connection is allowed.

SlySven, 2016-10-30:
A useful Q&A: [Why do daemons fork](http://stackoverflow.com/questions/4192472/why-do-daemons-fork#4192511) is on Stack Exchange.

it is so that the daemon IS completely independent of the parent that started it (otherwise it perhaps  _isn't_ a daemon?) and it ought to do it twice so it isn't a "process leader" :confused: - though I think I need to read a bit more deeply on the significance of that...

jdiamond, 2016-10-30:
I only meant to reference the first section in that link. I don't believe the section on TCP/IP is relevant to pigpiod.

I submitted a PR for an option to do this. It's not much more than an `if` statement around the code that does the forking in `main()`.

Current users shouldn't be affected and shouldn't have to learn anything new because the default usage doesn't change. Run `pigpiod` and it forks so you get your shell back, but you have to use `kill` to terminate it.

Run `pigpiod -g` and it stays in the foreground so you can just use Ctrl+C to terminate it, but you lose the usage of your shell while it's running.

When run in a container with `-g`, it will be PID 1 and will be the only process Docker has to monitor. Tiny init systems like dumb-init should also be able to monitor it.

The Raspbian package could also use this to avoid needing to set `Type=forking`. The default behavior of systemd is to assume services don't fork and it's how the systemd man pages recommend writing service files, if possible.

closed 2016-11-01 by joan2937