joan2937/pigpio #96: New Function EVM and EVT

Joey-1970, 2016-11-07:
Hello Joan,

here I'm again... ;-)

Please, can you explain me, what this new function are good for? I'm not sure, if I understand this right...
Is it something like the notification?
Set it with EVM (115) with the GPIO you want to monitor, and get back information with EVT (116) when soemthing happend on this GPIO?

Joachim

joan2937, 2016-11-08:
They are simply a convenience.  I had to add an event to cater for the BSC I2C/SPI slave peripheral so I decided I might as well generalise the functionality.  To generalise needed about 10% of the total effort so it seemed silly not to do so.

You can use an event to inform other interested parties listening to that Pi that something has happened they need to be aware of.  There is no more to it than that.  I expect there is little demand for such a function as I've not needed the function myself until I was implementing the BSC I2C/SPI slave.

Joey-1970, 2016-11-08:
Thanks for your answer!

So I think I actually can't use it in my project.

Joachim

closed 2016-11-08 by Joey-1970