joan2937/pigpio #97: Raspi Camera

Cabalist, 2016-11-17:
I hesitated to write this as it seems to not be covered by this projects goals but I am curious if there are any plans to support the Camera pins/module in the future?

joan2937, 2016-11-17:
Do you have anything [particular in mind?

I don't have a Pi camera and have never given this area any thought.

Cabalist, 2016-11-17:
Want one?  I'd happily send you one.

I love the daemonization of the GPIO pins.  I am able to develop and test on my machine without having to deploy to a Raspberry Pi every code change.  Being able to control multiple Pis from a central system has also been a huge benefit to me.  

I'd be interested in seeing something similar where I can call code that connects to a daemonized camera module running on a raspberry pi.

joan2937, 2016-11-17:
You may be able to achieve what you want now.

You can launch a shell script with arbitrary parameters.

See pigs [shell](http://abyz.co.uk/rpi/pigpio/pigs.html#SHELL), Python [shell](http://abyz.co.uk/rpi/pigpio/python.html#shell), etc.

Cabalist, 2016-11-17:
Neat.  I was unaware of this functionality in pigpio.  Thanks for your response.  This solves my problem and I think you can consider this closed.

You made something I really appreciate.  Thanks!

closed 2016-11-17 by joan2937