joan2937/pigpio #100: bbI2CZip: size of the output buffer

tnogueira, 2016-12-20:
Given the check on the outLen:

https://github.com/joan2937/pigpio/blob/master/pigpio.c#L10351

the outBuf must be at least 1 byte larger than the number of bytes we actually want to read. This makes it a bit counterintuitive/unexpected for someone using the function.

Should it maybe read **_if ((bytes + outPos) <= outLen)_**   ?

The same applies to PI_I2C_WRITE.

joan2937, 2016-12-21:
You are right. I will change lines 10351 and 10376 as you suggest for the next version (V59).

I have no timescale for that release at the moment.

tnogueira, 2016-12-22:
Thanks for the quick reply to both issues, and keep up the great work with pigpio!

Thanks again

joan2937, 2016-12-22:
Should be fixed in the now released V59.

joan2937, 2017-01-05:
Assumed fixed.

closed 2017-01-05 by joan2937