joan2937/pigpio #101: Publish on PyPI

bennuttall, 2016-12-20:
Hi @joan2937 - would you mind if I was to publish this on the Python Package Index so that it can be installed with pip?

If you want to do it yourself - please do - but if not, I'm happy to, and I'll try to keep it updated as you do releases.

Thanks

joan2937, 2016-12-20:
Thanks, that would be useful, I'm quite happy for you to publish the Python package on the Python Package Index.  Let me know if I need to do anything.

bennuttall, 2016-12-21:
Thanks! It's now published on PyPI and can be installed with pip. I've sent a PR (#103) with some small changes to setup.py.

bennuttall, 2016-12-21:
Not working currently, but I'm looking into it...

closed 2016-12-21 by joan2937

MrYsLab, 2017-01-12:
On pypi, the version is shown as 1.35. How does that correlate to V.60?

joan2937, 2017-01-12:
pigpio is made up of a number of components.  Each component has its own version number.

Every now and then I roll out a consistent set of components and bump the overall version number by 1 (the current number 60).

The latest pigpio.py component (the Python module) is currently version 1.35 as described by its VERSION string.  Click on https://github.com/joan2937/pigpio/blob/master/pigpio.py to see the current version source.

As far as possible I try not to invalidate earlier versions of software when any other components are updated.

MrYsLab, 2017-01-12:
First, at the outset, let me say that I love pigpio so I am not trying to be critical. I will continue to recompile and reinstall from source for updates, I just want to understand what the purpose and scope of the pypi distribution is.

That said, I understand that you are giving individual components version numbers. No problem there. Also, if my understanding is correct that the full set of components yields the library version. Some components may not have been touched for a while, and some may be new. The version number I see when executing get_pigpio_version() the value returned is the collective version of all the components.

Before and after running
```
sudo pip install pigpio --ugrade
```
When I run this script, the version number I see is 50. But this clearly out of sync, since the latest version of pigpio.py, version 1.35 is marked with V59.

Now on to my confusion. I do not understand the purpose of the pypi distribution and its being associated with the component number. From a user perspective, all I care about is the version number (I fully understand why the component version would be meaningful and useful to you). I have several PIs floating around and I do not update pigpio for all of them at the same time, but I do care, when using a given RPi, what version I am running.

So, why would I ever use pypi and not grab the tar.gz and rebuild? What is the use case for the pypi distro?

Is the pypi distribution only reflecting and installing the single pigpio.py file? If so, I would prefer that as a release, the name on pypi be something like pypigpio. 

I use other libraries that are C/C++ based and that have a python wrapper. ZeroMQ is a case in point. After compiling and installing ZeroMQ, I then get the python wrapper from pip using the package name of "pyzmq". 

Sorry to take up so much of your time, but I truly think this dual numbering scheme will result in others being confused as well.

bennuttall, 2017-01-12:
Use case:

- I am running the pigpio daemon on my Raspberry Pi
- I have the pigpio Python library installed on my PC
- I am writing Python code on my PC which controls the GPIO pins on my Raspberry Pi

How did I get there?

On the Pi:

- I installed the pigpio C library and ran the daemon

On my PC:

- I installed the pigpio Python library

The pigpio C library does not need to be installed on the machine with the Python library installed, as it's not actually doing any GPIO stuff. It just needs to know how to communicate with the Pi that's doing GPIO.

The reason I wanted to publish pigpio on PyPI is that I maintain the [gpiozero](http://gpiozero.readthedocs.io) Python library, a friendly API designed for use in education. Easy installation makes physical computing more accessible to more people.

I managed to get both pigpio (C) and pigpio (Python) into Raspbian (there are debian packages in the repo, and they're installed by default), and that made it work out of the box for Raspberry Pi users. In order to use pigpio's remote GPIO control from a PC, you just need to install the pigpio Python library, and I wanted to make that as simple as possible. In [x86 PIXEL](https://www.raspberrypi.org/blog/pixel-pc-mac/) it's installed by default. In Linux, Mac and Windows all you need to do is install pip and then pip install pigpio, rather than git clone and build from source.

MrYsLab, 2017-01-12:
Totally understood - I forgot about this feature of the library. Didn't mean to waste your time, thanks for answering.

joan2937, 2017-01-12:
That said I would admit that pigpio version numbering is a mess.

I have implemented all I wanted to implement.  Any future changes are more than likely to be bug fixes.

I'll have a look around and see if I can figure out some sort of versioning solution.

MrYsLab, 2017-01-12:
Not sure if this helps, but I find using the Github release feature very useful. You can take a look at one of my project's release pages here: https://github.com/MrYsLab/pymata-aio/releases

It makes it easy to pick an earlier release if needed, and you could still have a version number embedded in each component. 

Again, thanks for a great library

commit 4c039aa2 (2016-12-21): Merge pull request #103 from bennuttall/master

Python packaging changes, close #101