joan2937/pigpio #102: PWM output can glitch when updating duty cycle

jbeale1, 2016-12-20:
I am puzzled by the output of a PWM test using PIGPIO. WIth the code below, as expected, the duty cycle of the pulse moves smoothly up and down EXCEPT, as not expected, sometimes it stays high for one full 10 msec period, then immediately returns to the normal short duty cycle. The problem is intermittent, it happens maybe once in 2000 PWM cycles (20 seconds at 100 Hz). I suspect it happens when the PWM frequency is changed with pi.set_PWM_dutycycle() at just the wrong moment relative to the phase of the PWM trigger. Can I avoid this?

Scope Trace: https://goo.gl/photos/jjTCGfjfmibjQJEX7
[image]
Python code generating the problem: 
import time
import pigpio

LED_PIN = 25
pi = pigpio.pi()
pi.set_mode(LED_PIN, pigpio.OUTPUT)
pi.set_PWM_range(LED_PIN, 255)
pi.set_PWM_frequency(LED_PIN, 100)

while True:
    for i in range(5, 100, 5):
        pi.set_PWM_dutycycle(LED_PIN,i);time.sleep(0.1)
    for i in range(100, 5, -5):
        pi.set_PWM_dutycycle(LED_PIN,i);time.sleep(0.1)

pi.set_mode(LED_PIN, pigpio.INPUT)
pi.stop()

joan2937, 2016-12-21:
I tested with your script.

```py
import time
import pigpio

LED_PIN = 25
pi = pigpio.pi()
pi.set_mode(LED_PIN, pigpio.OUTPUT)
pi.set_PWM_range(LED_PIN, 255)
pi.set_PWM_frequency(LED_PIN, 100)

while True:
    for i in range(5, 100, 5):
        pi.set_PWM_dutycycle(LED_PIN,i);time.sleep(0.1)

    for i in range(100, 5, -5):
        pi.set_PWM_dutycycle(LED_PIN,i);time.sleep(0.1)

pi.set_mode(LED_PIN, pigpio.INPUT)
pi.stop()
```
Okay, this behaviour goes back to at least V16 (12-Jun-2014) so it looks like it's a feature rather than an introduced bug. Earlier versions are incompatible with the script.

That's a pity as at least an introduced bug suggests a fix. I'll look at the code to see if there is a (not too hard) way to get better behaviour.

joan2937, 2016-12-22:
Should be fixed in the now released V59.

jbeale1, 2016-12-22:
>  -------Original Message-------

Sounds great!  I am having trouble with the zip file though, I can download it, but cannot unzip it. File is 6934 bytes.
End-of-central-directory signature not found. ...
unzip: cannot find zipfile directory in one of pigpio.zip or pigpio.zip.zip and cannot find pigpio.zip.ZIP, period.

jbeale1, 2016-12-22:
nevermind my earlier message. Looked at my ".zip" file and it turns out to be an error HTML page generated from the 
wifi connection I used that wanted a click-through signin. Sorry!