joan2937/pigpio #104: Trouble starting up

GerardWassink, 2016-12-28:
Hi all,

On my Raspberry Pi I run the new Raspbian Jessie (with PIXEL). I run it headless through ssh, so I just run multi-user, not graphical.

Now I just downloaded the pigpio package using the recipe at [this page](http://abyz.co.uk/rpi/pigpio/download.html). So I did:

cd
wget abyz.co.uk/rpi/pigpio/pigpio.tar
tar xf pigpio.tar
cd PIGPIO
make -j4
sudo make install

So far, so good... 

So now I started pigpiod:
`sudo pigpiod`
no errors, and I can see that it's running:

pi@Columbus:~/PIGPIO $ ps alx | grep gpio
5     0 17784     1  20   0   9904  1684 -      SLsl ?          0:01 pigpiod
0  1000 17793   884  20   0   4276  1824 pipe_w S+   pts/0      0:00 grep --color=auto gpio
pi@Columbus:~/PIGPIO $

Now for the big test:

pi@Columbus:~/PIGPIO $ python
Python 2.7.9 (default, Sep 17 2016, 20:26:04) 
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import pigpio
>>> pigpio.start()
**Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'start'**
>>> exit()
pi@Columbus:~/PIGPIO $

Also this test fails:
`pi@Columbus:~/PIGPIO $ sudo ./x_pigpio # check C I/F
**_2016-12-28 14:14:21 initInitialise: Can't lock /var/run/pigpio.pid
pigpio initialisation failed._**
pi@Columbus:~/PIGPIO $ `

Unfortunately, as you can see it gives me errors.
Any clues?

Gerard

joan2937, 2016-12-28:
The pigpio module start() function was deprecated several years ago.

See issue #84.

closed 2016-12-28 by joan2937

GerardWassink, 2016-12-28:
Seems that I was looking at an old page on the Raspberry Pi site. 
Sorry for your trouble!

Gerard