joan2937/pigpio #107: Compiling on alpine

exelban, 2017-01-10:
Someone tried to compile library for alpine linux?
I receive next error:
pigpio.c: In function 'fileApprove':
pigpio.c:12678:15: warning: 'mperm' may be used uninitialized in this function [-Wmaybe-uninitialized]
       switch (toupper(mperm))
               ^
gcc -shared -o libpigpio.so pigpio.o command.o
strip --strip-unneeded libpigpio.so
size     libpigpio.so
   text	   data	    bss	    dec	    hex	filename
 236711	   5332	 596552	 838595	  ccbc3	libpigpio.so
gcc -o x_pigpio x_pigpio.o -L. -lpigpio -pthread -lrt
gcc -o pigpiod pigpiod.o -L. -lpigpio -pthread -lrt
strip pigpiod
install -m 0755 -d                /opt/pigpio/cgi
install -m 0755 -d                /usr/local/include
install -m 0644 pigpio.h          /usr/local/include
install -m 0644 pigpiod_if.h      /usr/local/include
install -m 0644 pigpiod_if2.h     /usr/local/include
install -m 0755 -d                /usr/local/lib
install -m 0755 libpigpio.so      /usr/local/lib
install -m 0755 libpigpiod_if.so  /usr/local/lib
install -m 0755 libpigpiod_if2.so /usr/local/lib
install -m 0755 -d                /usr/local/bin
install -m 0755 pig2vcd           /usr/local/bin
install -m 0755 pigpiod           /usr/local/bin
install -m 0755 pigs              /usr/local/bin
if which python2; then python2 setup.py install; fi
if which python3; then python3 setup.py install; fi
/usr/bin/python3
running install
running build
running build_py
creating build
creating build/lib
copying pigpio.py -> build/lib
running install_lib
copying build/lib/pigpio.py -> /usr/lib/python3.5/site-packages
byte-compiling /usr/lib/python3.5/site-packages/pigpio.py to pigpio.cpython-35.pyc
running install_egg_info
Writing /usr/lib/python3.5/site-packages/pigpio-1.35-py3.5.egg-info
install -m 0755 -d                /usr/local/man/man1
install -m 0644 *.1               /usr/local/man/man1
install -m 0755 -d                /usr/local/man/man3
install -m 0644 *.3               /usr/local/man/man3
ldconfig
Makefile:80: recipe for target 'install' failed
make: *** [install] Error 1

closed 2017-01-13 by joan2937

zinen, 2020-06-14:
I agree, I get the same problem when compiling on alpine. I found an explanation on [stackoverflow](https://stackoverflow.com/questions/36990951/ldconfig-seems-no-functional-under-alpine-3-3). The gist of it is that alpine can't use `ldconfig`.

**Solution for me:**
Did a fork of this repo and removed `ldconfig` from Makefile (removed line 112-114 and 134-136) that solved the issue. My fork here 9c18aaa4143366b978bf76e93702363d8f3c8244

whardier, 2021-02-16:
You might have luck aliasing ldconfig beforehand

zinen, 2021-02-17:
> You might have luck aliasing ldconfig beforehand

I guess you are right @whardier an alias would work too. Like running this before the make command:

 `alias ldconfig="echo ldconfig command disabled"`