joan2937/pigpio #109: Use of TCA9548A with PIGPIO

Joey-1970, 2017-01-18:
Hello Joan,

is it possible to use a TCA9548a I²C with PIGPIO?
How should I get the handle from Devices with the same Device Adress on different channels?

You write: "Physically buses 0 and 1 are available on the Pi. Higher numbered buses will be available if a kernel supported bus multiplexor is being used." - is this still valid?

Joachim

joan2937, 2017-01-18:
I guess that is a typo for the PCA9548 I2C multiplexor.

If so, yes by design it should work with pigpio,.

There is the multiple bus way with device tree where you add `dtoverlay=i2c-mux,pca9548=1` to `/boot/config.txt` and reboot.  That will give additional buses /dev/i2c-3 through /dev/i2c10 where you access each device using its unique bus.
```
Name:   i2c-mux
Info:   Adds support for a number of I2C bus multiplexers on i2c_arm
Load:   dtoverlay=i2c-mux,<param>=<val>
Params: pca9542                 Select the NXP PCA9542 device

        pca9545                 Select the NXP PCA9545 device

        pca9548                 Select the NXP PCA9548 device

        addr                    Change I2C address of the device (default 0x70)
```
Personally I see no real advantage doing that.  I'd just use the multiplexor at address 0x70 on bus `/dev/i2c-1` and select the device I want by writing its bit to 0x70.

Joey-1970, 2017-01-19:
Hello Joan,

I read here that there is an new syntax:
https://www.raspberrypi.org/forums/viewtopic.php?f=44&t=141517

dtoverlay=i2c-mux,pca9548,addr=0x70

Thanks a lot, because there are now a lot new users of your fantastic PIGPIO

Joachim

closed 2017-02-20 by joan2937

Joey-1970, 2017-03-10:
Hello Joan,

I think about this sentence:
"Personally I see no real advantage doing that. I'd just use the multiplexor at address 0x70 on bus /dev/i2c-1 and select the device I want by writing its bit to 0x70."

For example: If I switch on 0x70 to channel 1, how can I read from device address 0x20? Witch I2C-Bus I must use for the command?

Joachim

Joey-1970, 2017-03-10:
...when I use the way with changes in config.txt, I get more I2C-Bus (3 to 10). It is easy to program with pigpio, but the user must do this changes.
In your way, the user must not do this changes, but I don't know how to read/write the devices behind the Multiplexer...

Joachim

electrokean, 2017-03-11:
@Joey-1970 If you don't tell Linux about the I2C multiplexer, then you handle it yourself.  You then always use the one Linux i2c device for all communications, but you send commands to the multiplexer (@ 0x70) before addressing any other device to tell the multiplexer which of its multiplexed buses to connect through to the CPU.
Essentially you're taking control of the bus selection instead of the kernel driver.  The multiplexer slows things down a little either way due to the extra commands you need to send it.

Joey-1970, 2017-03-11:
Thanks for your answer!

For my understanding: If I want to use the raspberry own bus 1, I switch "off" the multiplexer and try to get a handle on bus 1 and (for example) address 20, if I want to use a device with the same device address behind the multiplexer channel 1, I switch the multiplexer to channel 1 and try to get a handle for the device with the same address again on bus 1?

Joachim

electrokean, 2017-03-11:
Lets say you have the mux at 0x70 and 4 multiplexed buses, each with a device at 0x20.
I would just get one handle for the mux, set it to select first bus, and then get a handle for the device at 0x20 on the first bus.
I can then send commands to the mux and any of the 4 devices, depending on the mux bus selection.
I haven't tried it, but I believe that would work - you'll need to keep track of the state of each device, but Linux and pigpio don't know or care much about the mux.
If you have other devices on some/all of the buses, you'd also need to take care of setting the mux prior to talking to them as well.

electrokean, 2017-03-11:
Note that you can't have a device of the same address before the mux as any "behind" the mux.
Nor can you switch off the mux (I guess that depends on the part used).

Joey-1970, 2017-03-14:
Hello Kean,

I need a little time - but it work!
Many thanks!!!

Joachim

electrokean, 2017-03-15:
Glad to hear that it worked