joan2937/pigpio #110: Pigpio can't keep the baud rate

RodrigoLGuimaraes, 2017-01-18:
I'm working on a project where I have a python application that uses pigpio to communicate using the MDB 4.2 protocol. It is a serial protocol that uses 9-bit words with 9600baud rate.

I did a bunch of tests running my python script and it works perfectly, both sending and receiving messages. My problem is that, when I start a bunch of other software that are supposed to run on my raspberry pi in conjunction with this project, the data that I send is not being read correctly on the other side. The data I read from the peripheral is read correctly though.

This is an example word that I'm sending when only the python script is running:

http://i.imgur.com/QfRisrQ.jpg

And this is the same example word being send when the python script is being run alongside a bunch of other programs:

http://i.imgur.com/9g9bRBP.jpg

I really think it might have to do with the lack of resources on the raspberry pi to process everything at the same time, but giving pigpiod a -20 process niceness(high priority) did not solve my problem. 

I then tried increasing the baud rate a bit to see if it worked, and I got it to work by using 9600 when it is running by itself and 10560 when it is running alongside other CPU-consuming software. I know it is not a good approach and I wanted to know if you guys have any idea on how to have pigpio keeping a stable 9600 baud rate.

Thanks.

RodrigoLGuimaraes, 2017-01-18:
Just found out that the software that is raising the issues is mpg321. I'll investigate the issue in the following days and will update here if I find a way to use mpg321 together with pigpio without issues.

joan2937, 2017-01-18:
Sound on the Pi (using PWM and/or PCM) and pigpio waves (used by pigpio to send software serial data) will not mix.

When using waves pigpio requires the use of both the PWM and PCM hardware peripherals of the Pi.  These are also the peripherals used by most sound applications.

RodrigoLGuimaraes, 2017-01-19:
Any suggestion on playing sound while using pigpio? I did a quick search online but couldn't find a way of doing it without using PWM nor PCM.

joan2937, 2017-01-19:
I don't use sound on the Pi and don't know of any answer.

Normally one of PWM or PCM will be available for use as pigpio can be told to use the other.

However if the [hardware PWM](http://abyz.co.uk/rpi/pigpio/cif.html#gpioHardwarePWM) function or waves are in use neither PWM nor PCM will be available for audio.

electrokean, 2017-01-19:
Maybe via a USB audio adapter?  If you can find drivers...

jam3sward, 2017-01-19:
Also, there are quite a few audio DACs available for the Pi, such as the "pHAT DAC" from Pimoroni and "Pi-DAC" from IQaudIO, also Wolfson Microelectronics.

SlySven, 2017-01-20:
:thought_balloon:  [Wolfson Microelectronics](https://en.wikipedia.org/wiki/Wolfson_Microelectronics#Products) were acquired by Cirrus Logic in 2014 and since then the [original card](https://www.element14.com/community/community/raspberry-pi/raspberry-pi-accessories/wolfson_pi) for use with the RPi B Rev. 2 (__with__ the P5 header) has been supplemented by the [Cirrus Logic Audio Card](https://www.element14.com/community/community/raspberry-pi/raspberry-pi-accessories/cirrus_logic_audio_card?ICID=wolfson-cirruslogic-section) for the RPi B+/2B/3B...

RodrigoLGuimaraes, 2017-01-25:
All great suggestions, but I really hope to be able to have audio on my pi without having to buy any extra parts, since this is to be bundled on a product, and having another part will make the final product more expensive. I don't need high quality audio, any idea on playing sound alongside pigpio without having to buy extra hardware?

RodrigoLGuimaraes, 2017-02-17:
Hello guys, was finally able to set this issue by buying an external USB sound card. Worked like a charm. Thanks for all the help.

closed 2017-02-17 by RodrigoLGuimaraes