joan2937/pigpio #111: Problem with handle > 31 for I2C

Joey-1970, 2017-01-19:
Hello Joan,

I get an error if the ask for a free I2C-Handle ist >31.
pi@RPiTest:~ $ pigs i2co 6 0x20 0
30
pi@RPiTest:~ $ pigs i2co 6 0x20 0
31
pi@RPiTest:~ $ pigs i2co 6 0x20 0
-24
ERROR: no handle available

It's same when I try it over the socket interface...

Is there something wrong?

Joachim

joan2937, 2017-01-19:
Nothing wrong, that is by design.

pigpio.h has

#define PI_I2C_SLOTS 32

Like all limits I didn't expect it to be reached.

Do you actually need more than 32 handles?

Joey-1970, 2017-01-19:
Hello Joan,

if it would be possible, I would be lucky if cound make it greater. So of the users of my module to connect IP-Symcon over your PIGPIO want build big networks of I²C-devices, so it is maybe better... ;-)

Joachim

joan2937, 2017-02-20:
In V61 the number of I2C handles has been increased to 64.

Hopefully this will be enough for most people.

I am reluctant to set an open (unbounded) number of handles but will consider doing so if I am convinced it is useful.  If you think you have a case please reopen this issue.

closed 2017-02-20 by joan2937

bludin, 2019-09-24:
Is there a compelling reason for limiting the maximum number of I2C-handles to 64 and if not, could it be increased (e.g. to 512)? I have a situation where this becomes limiting (a modular construct where each of the up to 48 modules with up to 10 I2C-devices receives its own I2C-bus via 6 TCA9548 muxes). I can program my way around this limitation but it would be nicer if I didn't have to

Beat

PS: Thanks for pigpio which is just wonderful!
PS2: sorry for double-posting, but my original inquiry didn't get an answer for several weeks and I suspect that's because it's considered misplaced given the existence of this thread

joan2937, 2019-09-25:
Only a few (two?) people have ever requested more handles.

Unless you are using pigpio remotely you probably oughtn't to be using the pigpio I2C functions.  They are mainly just a wrapper around the underlying I2C/SMBus calls.  Why not just use the calls directly.

Anyhow it is simplest if you just change the #define in your own copy of pigpio and rebuild.  There is not going to be a flurry or new releases causing problems.

bludin, 2019-09-30:
> Unless you are using pigpio remotely you probably oughtn't to be using the pigpio I2C functions. They are mainly just a wrapper around the underlying I2C/SMBus calls. Why not just use the calls directly.
Ah... thanks. I hadn't been aware of that.

immiz, 2019-12-08:
The error message can also occur when you don't close the handle correctly.

you can close it like:
`for ((i=0; i<32; i++)); do pigs i2cc $i; done`

or another clean way:
```
from pigpio import pi
x_pi = pi('192.168.0.51')
if not x_pi.connected:
    print('Unable to connect to RPi')
else:
    h = x_pi.i2c_open(1, 118)
    x_pi.i2c_close(h)
    print(h)
    if h > 30:
        x = h - 1
        while x >= 0:
            x_pi.i2c_close(x)
            x = x - 1
        print(str(h) + ' pigpio handles closed')
    x_pi.stop()
```

source: https://www.raspberrypi.org/forums/viewtopic.php?t=88545