joan2937/pigpio #114: Software Serial Transmission Does not work

efbick, 2017-02-16:
I am attempting to create a software serial connection using wave_add_serial to send the data and bb_serial_read_open to read data being received. I am able to receive the data perfectly but when I send "Hello World" the reception on the other side is completely different asci characters. I put the code below and the response in HEX. It seems like the bit order is changing some how but I am not sure how it is. Can't seem to find a pattern.

`import pigpio
import time
import sys
import numpy
import struct

txPin=20
rxPin=21

serialpi=pigpio.pi()
serialpi.set_mode(rxPin,pigpio.INPUT)
serialpi.set_mode(txPin,pigpio.OUTPUT)

pigpio.exceptions = False
serialpi.bb_serial_read_close(rxPin)
pigpio.exceptions = True

baudrate =9600

serialpi.bb_serial_read_open(rxPin,baudrate,8)
         
def Sendline(serialpi):
    serialpi.wave_clear()
    serialpi.wave_add_serial(txPin,baudrate,b'Hello world',0,8,2)
    wid=serialpi.wave_create()
    serialpi.wave_send_once(wid)
    while serialpi.wave_tx_busy():
        pass
    serialpi.wave_delete(wid)   
 
#while True:     
Sendline(serialpi) 
print 'sent'
time.sleep(.5)           `

The response on the other end is

`BA 02 00 AB DB D2 52 A4 8B 14 DA 74 BA 02 00 AB D3 D2 52 A4 8B 14 DA 3A BA 02 00`

joan2937, 2017-02-16:
I suggest there is a problem at the "other end" whatever that might be.

The script works fine when I connect a wire between GPIO 20 (pin 38) and GPIO 15 (pin 10 UART RXD) and cat /dev/serial0.

```
$ cat /dev/serial0
Hello world
Hello world
Hello world
Hello world
```
Note, I changed the script to add a \r\n to the string for testing and I have enabled serial on my Pi3.

efbick, 2017-02-16:
Thanks for the quick response. So if I plug the Tx directly into the serial RX on the pi, everything works fine just like you. Any other computer I use does not however. I've tried multiple computers with different usb to db9 connectors. I've tried different cables and different lengths. I've tried lowering the baud rate to 600. Any ideas?

joan2937, 2017-02-16:
The Pi's GPIO are 3V3 TTL.

What wiring are you using between the Pi and the other end?

What is it connected to at the other end?

efbick, 2017-02-16:
Ok I think I know where you going at here. Im sending 3v3 TLL on the pi and receiving 5V TLL on the computer. Didn't even think about that. I'll try to convert it and see if it works. Thanks

joan2937, 2017-02-20:
I am going to close this issue.  Please re-open if there does prove to be a software problem.

closed 2017-02-20 by joan2937