joan2937/pigpio #116: rawWaveGetOut depprecated but rawWaveGetOOL undefined

agnunez, 2017-02-23:
Hi, thanks for excellent pigpio.
I am using it to read 8 MCP3304 chips SPI's simultaneusly without problem (25000 sps by 8 channels). While recoding your demo, I saw that rawWaveGetOut was deprecated and suggestioin to be replaced by rawWaveGetOOL. Tried and got undefined at compile time. I installed pigpio from Raspbian package and updated. Any recommendation either to keep old function or integrate new one?

Thanks
Agustin

joan2937, 2017-02-23:
Have a look at the MCP3202 example http://abyz.co.uk/rpi/pigpio/examples.html#C_rawMCP3202_c

Hopefully that will be similar to the MCP3304.

agnunez, 2017-02-23:
Thanks, this is the code I took as source to adapt to MCP3304 and mine is working with rawWebGetOut on line 84 on your example. My question is if I triy to follow your recommenation, on pigpio website library function explation, to substitute rawWebGetOut by rawWebGetOOL, as you say former is deprecated, it stop working and do not compile throwing an error. I will leave as it is for the time being and works. 
Thanks

joan2937, 2017-02-23:
http://abyz.co.uk/rpi/pigpio/cif.html#rawWaveGetOut
http://abyz.co.uk/rpi/pigpio/cif.html#rawWaveGetOOL

In this case the functions have identical code.

In the examples I should have used rawWaveGetOOL.  I am unlikely to actually ever remove the deprecated rawWaveGetOut so I shall not bother to update the example at this stage.

I am baffled as to why the symbol rawWaveGetOOL is not present in the library so the code will not link.  I can reproduce that error.  If and when I discover the problem I'll update this issue.

joan2937, 2017-02-23:
I had an old raspberrypi.prg image of the library in my default search path.

By default pigpio puts the libraries in /usr/local/lib.

raspberrypi.org places the libraries in /usr/lib.

If you update the library from abyz.co.uk or from github it is best to delete the following old entries.

/usr/lib/libpigpiod_if2.so
/usr/lib/libpigpiod_if.so
/usr/lib/libpigpio.so

closed 2017-02-23 by joan2937

agnunez, 2017-02-23:
Thanks! I will do!

Agustin