joan2937/pigpio #119: localhost socket only not working

dplasa, 2017-02-26:
If I start with the localhost socket only option
`$ sudo /usr/bin/pigpiod -l`

x_pigpiod_if2 is unable to connect to the daemon:

```
$ ./x_pigpiod_if2
pigpio initialisation failed (-2003).
```
It will only work without the -l option.
I'm using the V61 on a RaspberryPi B
```
$ sudo /usr/bin/pigpiod -v
61

```

joan2937, 2017-02-26:
I will modify the start of pigpio_start() in pigpiod_if2.c as follows:
```C
int pigpio_start(char *addrStr, char *portStr)
{
   int pi;
   int *userdata;

   if ((!addrStr) || (strlen(addrStr) == 0))
   {
      addrStr = "localhost";
   }
```
If you get a chance could you try the above fix.

dplasa, 2017-02-27:
Hey, great! That did fix it for me!

joan2937, 2017-02-27:
Okay, I'll add that change in the next software release.  I am not sure when that will be.

joan2937, 2017-04-24:
Should be fixed in V62.

closed 2017-04-24 by joan2937