joan2937/pigpio #120: SPI BitBang sample having "undefined references"

paolopugay, 2017-02-28:
Hi Joan,

As I compile the sample program for Bit Bang SPI (from the library documentation in **bbSPIXfer**, http://abyz.co.uk/rpi/pigpio/cif.html#bbSPIXfer), I having the following undefined references.

Errors are as follows:
gcc -Wall -pthread -o "bbSPIx_test" "bbSPIx_test.c" -lpigpio (in directory: /home/pi/Desktop/comm_test)
/tmp/ccGfO5wm.o: In function `main':
bbSPIx_test.c:(.text+0x9c): undefined reference to `bbSPIOpen'
bbSPIx_test.c:(.text+0xc0): undefined reference to `bbSPIOpen'
bbSPIx_test.c:(.text+0xf4): undefined reference to `bbSPIXfer'
bbSPIx_test.c:(.text+0x120): undefined reference to `bbSPIXfer'
bbSPIx_test.c:(.text+0x154): undefined reference to `bbSPIXfer'
bbSPIx_test.c:(.text+0x1ac): undefined reference to `bbSPIClose'
bbSPIx_test.c:(.text+0x1b4): undefined reference to `bbSPIClose'
Compilation failed.
collect2: error: ld returned 1 exit status

I'd compile it in Terminal and Geany using RPI 3 running Raspbian GNU/Linux 8 (jessie).

Is the SPI Bit Bang using bbSPIXfer is still implemented in the latest version (61 v)?

Also, please see the attached screenshots and the sample program.
[comm_test.docx](https://github.com/joan2937/pigpio/files/806642/comm_test.docx)
[bbSPIx_test.txt](https://github.com/joan2937/pigpio/files/806644/bbSPIx_test.txt)

Thanks

joan2937, 2017-02-28:
I've just cut&pasted and it builds okay.

The most likely problem is you have an earlier version of the library earlier in the search path.

By default pigpio puts the libraries in /usr/local/lib.

raspberrypi.org places the libraries in /usr/lib.

If you update the library from abyz.co.uk or from github it is best to delete the following old entries.

/usr/lib/libpigpiod_if2.so
/usr/lib/libpigpiod_if.so
/usr/lib/libpigpio.so

paolopugay, 2017-02-28:
I'd follow the installation in method 1(in http://abyz.co.uk/rpi/pigpio/download.html) and file are created in the home folder (/home/pi/PIGPIO).

The issues have been resolved by removing the libpigpiod_if2.so, libpigpiod_if.so and libpigpio.so in the /usr/lib directory.

Thank you very much for the assistance :)

closed 2017-02-28 by paolopugay