joan2937/pigpio #121: Doesn't work on raspbsd

gczuczy, 2017-03-07:
Hello,

Could you please fix the build for FreeBSD? Apparently it assumes there's gcc (which is false, there's clang under the name "cc", per standards), and a couple of libs/includes seem to be different. As SPI is work-in-progress on freebsd, I guess a lot of people - including me - would love to use your lib for SPI communication.

Thanks in advance,

joan2937, 2017-03-07:
I can't help with that.  I don't use FreeBSD and have no idea of what would need to be done to properly build for that environment.

If someone knows of a method of making the build work for Debian/Raspbian as well as Arch and BSD I would love to know.

joan2937, 2018-02-08:
If anyone has any concrete proposals please reopen this issue.

closed 2018-02-08 by joan2937