joan2937/pigpio #122: Is gpioSetAlertFunc() thread based?

PTS93, 2017-03-14:
I have an infinite loop and wanted the callback function write to a variable when the monitored GPIO goes high. The loop would then alter its function whenever I receive a high.
Though I was wondering if that is a safe thing to do or do I need a mutex?

joan2937, 2017-03-16:
You shouldn't need a mutex.  The callbacks are made from a single thread so your callback will run in that threads context.  The callback will run to completion before the next call is made.

PTS93, 2017-03-16:
Great! So the thread where I register the callback function for a GPIO is the same thread the callback will run in? How does that work without me calling some sort of update function in my thread or blocking it? Just curious, have a hard time understanding how the function works by reading the relevant functions in pigpio.c.
Also is this true for all callback based functions?

fivdi, 2017-03-16:
I'd be more inclined to say it depends on the exact use case. If a single GPIO is being monitored and the callback is setting a single primitive variable like an int and alerts are not occurring at a high rate, then a mutex may not be needed. I'd feel better using one.

fivdi, 2017-03-16:
> Great! So the thread where I register the callback function for a GPIO is the same thread the callback will run in?

No, they are different threads.

joan2937, 2017-03-16:
There is a single sampling thread which calls all the needed callbacks in order.  I've always understood that all the callbacks are called in the context of the sampling thread.  The sampling thread will not generate another callback until the previous callback has completed.

fivdi, 2017-03-16:
Yes, there's the sampling thread which calls the callbacks in order in the context of the sampling thread and there's another thread with the infinite loop mentioned in the first post.

The sampling thread may call multiple callbacks in order in quick succession without the infinite loop thread being allocated any processing time.

PTS93, 2017-03-16:
Ah ok, that I can wrap my head around now. I will have multiple GPIOs monitored so a mutex seems to be in order. Any recommendations? My code is written in C++, would it be ok to just use std::mutex or do I need to use pthread_mutex? I have zero experience working with pthread directly.

joan2937, 2017-03-16:
I assumed that since the sampling thread makes a standard function call to the callback that callback was operating in the context of the sampling thread.

The spinning thread may not even have registered the callback.  I don't think a C function has any concept of ownership.  It simply knows the thread that it is called by.

fivdi, 2017-03-16:
In a similar scenario I used semaphores (sem_wait, sem_post, ...) but std::mutex should work too.

PTS93, 2017-03-16:
Thanks I wasn't aware of that concept, sounds useful.

closed 2017-03-16 by PTS93