joan2937/pigpio #123: Problem with Serieal read

Joey-1970, 2017-03-20:
Hello Joan,

I'm back again... ;-)

I have a serial Device but I don't get back data on the PIGPIO-socket-command 80 (SERR)...
I Can read with socket command 82 how much bytes are there, and I can read with this handle usings pigs (e.g. pigs serr 2 10), but when I use PIGPIO-socket-command 80 (SERR) I get back nothing...

I think my code runs until a PIGPIO-update, but I don't now which update...

Is this possible?

Joachim

Joey-1970, 2017-04-18:
Hello Joan,

long time without any feedback... :-(

Can you or someone else tell me, how I can turn back to an older PIGPIO-Version?

Joachim

joan2937, 2017-04-18:
You can find earlier versions in tar and zip form at http://abyz.co.uk/rpi/pigpio/VERSIONS/

I presume you can also use the standard git functions to extract earlier versions from the github repository.

joan2937, 2018-02-08:
I am not sure if this is an error in pigpio or user code.

Please reopen if you can provide a small script which demonstrates the error,

closed 2018-02-08 by joan2937