joan2937/pigpio #126: Wave deletion does not happen as expected

ingvardsen, 2017-04-24:
I must first say im impressed by this library. Im using it for a hobby CNC project and it makes it possible to do parallel execution of waves for several motors based on GPIO's.

But i think there is an issue with wave deletion.  
When using a python add-create-send-delete loop for waves, i cannot always remove the completed waves. It leads to 'No more CBs for waveform' when creating new waves.

I can recreate the problem with a simple loop. When the deletion does not happen, the wave id increases continuously.

and here is the code
```import pigpio
import time
import random

pi = pigpio.pi('navio2.local')

# system info
print 'pigpiod version: {}'.format(pi.get_pigpio_version())  # 61
print 'pigpio.py version: {}'.format(pigpio.VERSION)  # 1.36
print 'wave_get_max_pulses: {}'.format(pi.wave_get_max_pulses()) # 12000

gpio = [19, 5, 25]

total = 30000
blocksize = 3000
iter = total / blocksize

waves = []
wavepos = 0

"""
When running with different delay, the loop fails with pigpio.error: No more CBs 
for waveform'.

But when running with equal delays, the loop completes. 
Proberbly because the waves are reused in gpioWaveCreate.
"""
delays_that_fails = [200, 0]       # fails
equal_delay_does_not_fail = [200]  # to se an execution as expected

delay = delays_that_fails

for p in range(total / 6):

    waves.append(pigpio.pulse(1 << gpio[0], 0, random.choice(delay)))
    waves.append(pigpio.pulse(1 << gpio[1], 0, random.choice(delay)))
    waves.append(pigpio.pulse(1 << gpio[2], 0, random.choice(delay)))

    waves.append(pigpio.pulse(0, 1 << gpio[0], random.choice(delay)))
    waves.append(pigpio.pulse(0, 1 << gpio[1], random.choice(delay)))
    waves.append(pigpio.pulse(0, 1 << gpio[2], random.choice(delay)))

print 'pulses: {}'.format(len(waves))

pi.wave_clear()
iteration = 0
old_id = -1

# add-create-send-delete loop with a sleep delay after send
while wavepos <= len(waves) - blocksize:

    new_wavepos = wavepos + blocksize
    slice = waves[wavepos:new_wavepos]
    wavepos = new_wavepos

    iteration += 1
    print 'iteration %s' % iteration

    pi.wave_add_generic(slice)

    new_id = pi.wave_create()
    pi.wave_send_using_mode(new_id, pigpio.WAVE_MODE_ONE_SHOT_SYNC)

    while pi.wave_tx_at() != new_id:
        time.sleep(0.01)

    if old_id >= 0:
        print 'delete wave id %s' % old_id
        pi.wave_delete(old_id)

    old_id = new_id

pi.stop()

print 'complete'
```

ffleandro, 2017-08-06:
I can also replicate this issue.
wave_clear or wave_delete does not work as expected, not sure which one.

joan2937, 2017-08-07:
I can't remember why I just labelled this without comment.

Did I reply by e-mail?

When a wave is deleted it is simply marked for deletion.  Its resources will not be reused except under certain conditions.  One of these is that if a new wave is created needing exactly the same resources then it will reuse the old space.  It's safest to force all waves to use the same resources if you want to rapidly create/delete waves in a loop.

ffleandro, 2017-08-07:
I'm not the original poster, so can't comment on that.

Here is my scenario:

```
wave_clear()
wave_add_generic() #returns 1700 pulses for instance
wid = wave_create() 
wave_send_once(wid)
while wave_tx_busy():
    time.sleep(0.1)
wave_delete(wid)
```

If I run this code again, `wave_add_generic` will return 3400, 5100 on the next one, etc...

ingvardsen, 2017-08-08:
No i did not recieve a mail.

In my case i was able to release the waveform resources after it is deleted, when the amount of control blocks was the same as the previous. It was not deleted / reused if different.

The tricky part is that it's not only the number of pulses that gives the number of control blocks. The delay can  influence the number of cb's dependent on the length. So counting the pulses when building waves will not always work.

I think the issue here is that the dokumentation on wave_delete at least should have a note about the conditional deletion of the waves.

guymcswain, 2017-08-30:
Umm, this appears to be a real issue that needs to be addressed.  Try `waveAddSerial()` and you find,  unless you want to send the same message over and over, that you quickly run out of resources and can no longer transmit.  Is there something about the hardware that prevents `waveDelete()` from freeing the resources immediately?

guymcswain, 2017-08-30:
The source is the documentation:
```
int gpioWaveDelete(unsigned wave_id)
{
   DBG(DBG_USER, "wave id=%d", wave_id);
   CHECK_INITED;
   if ((wave_id >= waveOutCount) || waveInfo[wave_id].deleted)
      SOFT_ERROR(PI_BAD_WAVE_ID, "bad wave id (%d)", wave_id);
   waveInfo[wave_id].deleted = 1;

   if (wave_id == (waveOutCount-1))
   {
      /* top wave deleted, garbage collect any other deleted waves */
      while ((wave_id > 0) && (waveInfo[wave_id-1].deleted)) --wave_id;

      waveOutBotCB  = waveInfo[wave_id].botCB;
      waveOutBotOOL = waveInfo[wave_id].botOOL;
      waveOutTopOOL = waveInfo[wave_id].topOOL;
      waveOutCount = wave_id;
   }
   return 0;
}
```
So resources are only freed if the most recently created wave is deleted.  Ok, so now the issue is you can't create a new wave whilst the current wave is transmitting, send the new wave synchronously, then delete the wave just completed.  **It would be really nice if this worked!**

Also, `gpioWaveClear()` seems pretty worthless.  Is there not a way to re-initialize the wave generator without terminating pigpio?

joan2937, 2018-02-08:
I have clarified the documentation for wave delete.

I am not planning any code changes in this area, mainly because I can think of no way of doing it better than it is currently done.

Please reopen if you know of a workable method of better reusing deleted waves.

guymcswain, 2018-02-09:
@joan2937,

Thanks for looking into this and other issues in the pigpio backlog.  It has been a while but the use case I have is being able to transmit a long waveform by breaking it into smaller segments then transmitting the segments back-to-back to reproduce a continuous waveform at the output.  To do so, you need to create the next waveform segment, N, while the last segment, N-1, is transmitting.  After detecting the status of waveform segment N-1 being completed, you need to delete N-1 and then begin creation of waveform segment N+1.  As pigpio currently stands, wave delete removes the last (latest) wave, which in my example above is waveform N.

I understand that what I have described is the use case and not the solution.  I'm hoping that you have an inspirational moment and come up with one.

joan2937, 2018-02-09:
I wouldn't hold your breath waiting for a fresh inspiration from me.  I have given the area a considerable amount of thought.  Reusing resources boils down to shifting DMA control blocks around while they are actually being used.  It is too horrendous to contemplate.  Have a look at the wave chain code which allows loops to be repeated a specific number of times.  I couldn't write that again however hard I tried.

The best I can think of is making it easier to create waves of consistent size thus allowing their reuse.

nachoplus, 2018-06-26:
Hello,

I came across with this issue in my use case and I am looking how to solve. What does it means **consistent size** in this context? It is enough same length in micros and same number of pulses?

Lets put and example. Supose two wave of length 10 ms composes with pulses of 1 ms 

Wave1: off-off-on-on-off-off-on-on-off-off  
Wave2: off-on-off-on-off-on-off-on-off-on  

Are this two waves 'consistent in size' ?

Thanks

Jul3k, 2020-02-28:
> No i did not recieve a mail.
> 
> In my case i was able to release the waveform resources after it is deleted, when the amount of control blocks was the same as the previous. It was not deleted / reused if different.
> 
> The tricky part is that it's not only the number of pulses that gives the number of control blocks. The delay can influence the number of cb's dependent on the length. So counting the pulses when building waves will not always work.
> 
> I think the issue here is that the documentation on wave_delete at least should have a note about the conditional deletion of the waves.

@ingvardsen how did you match the number of used control blocks? I ran into the same issue and  looking at the source code it seems that not only the CBS need to match but also the OOL for a reuse of a deleted wave. Since the OOL can't be determined from the python interface I wonder what your approach was to get around this limitation?

As @joan2937 mentioned a easy way to create waves of consistent size would be a a nice addition here. Are there any updates on this?

ingvardsen, 2020-02-28:
@Jul3k i did not solve the problem. 
But in case you are working on a CNC, 3D print project or other realtime pulse generation from Python code, have a look at this [PyCNC](https://medium.com/iotforall/how-to-build-a-3d-printer-in-python-b05af32489f5). 
GitHub repos [here](https://github.com/Nikolay-Kha/PyCNC).

Jul3k, 2020-02-28:
@ingvardsen I do indeed work on project for synchronized stepper movement. I started with PyCNC but the problem is that it does not fit all the requirements. I found extending PyCNC difficult because it is already a very complete solution with everything included for a 3D Printer. The library I wrote is small and works actually quite well with step frequencies up to 100kHz for synchronized movements (using numpy) and variable number of drives . The problem is that I discovered the limitation of piogpio after most of the work was done already during testing of the code with low accelerations. I really hope @joan2937 has an idea on how to get around this limitation.

guymcswain, 2020-02-29:
@Jul3k , are you running the latest version?  The reason I ask is there was a fix to #223 which *may* have had some influence on this issue.

If you want to get attention on this issue you should reopen it.  Let me know if that is not possible since you are not the author.

Jul3k, 2020-02-29:
@guymcswain, yes I am running version 74 and I am not able to reopen this issue.

from `gpioWaveCreate`:

    if (waveInfo[i].deleted             &&
         (waveInfo[i].numCB   == numCB)   &&
         (waveInfo[i].numBOOL == numBOOL) &&
         (waveInfo[i].numTOOL == numTOOL))
      {
         /* Reuse the deleted waves resources. */
         wid = i;
         break;
      }

from `waveCBsOOLs`:

    numCB++;
    for (i=0; i<numWaves; i++)
    {
       if (waves[i].gpioOn)                 {numCB++; numBOOL++;}
       if (waves[i].gpioOff)                {numCB++; numBOOL++;}
       if (waves[i].flags & WAVE_FLAG_READ) {numCB++; numTOOL++;}
       if (waves[i].flags & WAVE_FLAG_TICK) {numCB++; numTOOL++;}

       numCB += waveDelayCBs(waves[i].usDelay);
    }

so is it correct that not only the number of CB but also the OOL need to match?

guymcswain, 2020-02-29:
> So is it correct that not only the number of CB but also the OOL need to match?

I don't currently know the answer.  But first, can we agree on the problem statement.  I propose it is this one:

*"Pigpio wave methods do not support the generation and transmission of continuous waves."*  Continuous wave(s) generation require a "create-sync-delete" loop that can run indefinitely without exhausting the CB or other DMA resources (within reason).  I think this covers the use case for the OP as well as my own.  

Please note that:

> an easy way to create waves of consistent size would be a nice addition here

would not resolve what I'm describing as a continuous wave.

Given your situation fits this problem statement, I'm willing to put in the time to investigate if and how it can be accomplished.  I believe this issue would then fit into the 'feature/enhancement' category as opposed to a fix.

Jul3k, 2020-02-29:
I agree that this is the general use case and I would say that this is also a use case which would make pigpio a very powerful tool (it is already). I would be happy if you could look into this and I am also willing to support.

My understanding from what @joan2937 said is that it will be quite difficult to implement a feature that frees resources from deleted waves directly.

Let me explain why I was asking for a way to create waves of consistent size:
There could be two waves that each require half of the available resources. One acts as a buffer which is filled while the other one is being transmitted. The transmitted wave is then deleted and overwritten with the newly created wave occupying the same amount of resources. The waves could be padded so that they always require the same amount of resources independent from the pulses which have originally been added to them. This could repeat indefinitely.

A way to delete a wave and free the resources directly would of corse be the more elegant and cleaner approach, which I would also prefer. I also see this as a `feature/enhancement`

guymcswain, 2020-02-29:
> There could be two waves that each require half of the available resources. One acts as a buffer which is filled while the other one is being transmitted.

I was thinking along the same line.  But I was also assuming that it would not make sense to force fit it into the existing regime but to make a new mode perhaps.  At the end of the day, it can't break existing functionality.

@joan2937 , if you could let us know if we're going down a bad path or give us other ideas to consider, we'll do the heavy lifting.

joan2937, 2020-02-29:
The ability to reuse a previous wave in-situ was an attempt to sole this problem.  Obviously an insufficient attempt.

Wave resources are allocated from several blocks of memory.  These blocks are easily fragmented as waves are created and deleted, especially if the resources required change.  You can not (rather I know of no way of doing so) move allocated blocks to unfragment the space.  **In theory I suppose you could build up a wave from non-contiguous memory.  I almost certainly would have done so if I had thought of this problem.**

My current idea was to pad out waves so they would use a consistent size, that is what I was going to do.  Perhaps it would be better now to address the root problem and think of a fresh method of allocation.

guymcswain, 2020-02-29:
Two waves padded out to consume half the DMA resources, while not necessarily elegant may be the right approach.

Jul3k, 2020-03-02:
Is there a way to pad the waves with the current version of PIGPIO?

guymcswain, 2020-03-02:
I'm still coming up to speed on the waveform generation code.  If you pad with a pulse of [0,0,0] pigpio will not consume a CB.  You could try to pad with [0,0,1] within your waveform to experiment.

Jul3k, 2020-03-02:
I had a similar problem and numpy and numba where a great help. Here is a link to some code I use for testing [Stepper Puls Generation](https://gist.github.com/Jul3k/6be814fc13e8c0e38ca8c4209d3e0c8c). Have a look at the wave_add_numpy_array function, this was a great step to speed up the generation process.

guymcswain, 2020-03-02:
I'll check that out later, I'm in information overload :)

The control blocks of the DMA are natural linked list structures.  While the CB resources are the critical factor in determining if a waveform 'can fit', there is this concept of OOL which are linear structures tied to the start and end of the CB chain.  Therefore the CB must be structured linearly in memory.  See this comment:
https://github.com/joan2937/pigpio/blob/ef48a043af0f54ecfcd8bcb2515c96ae1b13ccee/pigpio.h#L461-L472

guymcswain, 2020-03-02:
... so the result of your padding experiment would be of great interest.

Jul3k, 2020-03-02:
Ok, I tried what you suggested using [0,0,0] gives no change on the consumed CBs. Padding with [0,0,1] increases the consumed CBs but not by the number of added [0,0,1]. I do not see a pattern in the added CBs but the amount is always below the number of pulses added for padding

Jul3k, 2020-03-02:
https://github.com/joan2937/pigpio/blob/ef48a043af0f54ecfcd8bcb2515c96ae1b13ccee/pigpio.c#L2948-L2957
This might explain why no CBs are added for a zero delay

guymcswain, 2020-03-03:
I don't see debug statements in the internal wave related functions.  What specifically do you want to see?  I could compile a version with a custom function to return some of the internal data.  I'd rather wait though until I can get more understanding of the OOL data structure.

But it looks like your discovering some potentially useful information in your experiments!

guymcswain, 2020-03-03:
> I believe the OOLs also need to fit.

Yes, indeed both OOLs and CBs must match to reuse a wave id.  This makes it quite challenging.

guymcswain, 2020-03-03:
@joan2937 , you can unsubscribe to this issue and we will only pull you into the discussion when needed - as in now for this topic:

The waveform flags seem to only be accessible using the C/IF methods and are designated as "expert" and "not intended for general use".  In the waveform data structures, the flags run from top to bottom, counter to normal waveform constructs.  There is probably some mechanism that I'm not seeing to prevent writing the incoming 'level' and 'tick' data into the same buffer that has the outgoing OOLs.  Do I understand this correctly so far?

And my real question is who uses the flags or where are they used?  If they're not used, it may simplify the waveform data structures if they could be eliminated.  Your thoughts please.

Jul3k, 2020-03-03:
@guymcswain padding with [0,gpio,0] has an unwanted side effect as it adds an additional delay to the sequence. It kind of makes sense because also the DMA needs cycles to execute the copy. 1000 such pulses at the start of the wave result in an additional delay of around 240 us. So [0,gpio,0] is also not a good option for padding. Currently I don't see any more options to perform the padding from the python side. Hope you are having more luck ...

guymcswain, 2020-03-03:
I have a proposed solution that I'm going to code and test over the next few days.  `gpioCreateWave(size)` will allocate the number of CBs and OOLs for the wave by calling `wave2Cbs(...)`  with parameters that are sized accordingly.  `wave2Cbs()` will link the last CB to the end of the wave's allocated block.  All waves will have an exact count of CB and OOL so `waveCreate(size)` will always find a place as long as a wave got deleted.

I'm sure I'll uncover some problems but it looks like the basic structures in place are suitable for this change.  I'll push a 'feature branch' when I'm code complete and have begun testing so that you can do so in parallel.

`size` will be the proportion of the maximum resources.  ie, `size=4` would consume one quarter of the full resources available.  I believe your application would want `size=2`.

joan2937, 2020-03-03:
@guymcswain 

Are you referring to the following structure?
```c
#define WAVE_FLAG_READ  1
#define WAVE_FLAG_TICK  2

typedef struct
{
   uint32_t gpioOn;
   uint32_t gpioOff;
   uint32_t usDelay;
   uint32_t flags;
} rawWave_t;
```
The READ flag is only used by rawWaveAddSPI.

The TICK flag is only used by some example code (pps.c) which generates a pulse on the second.

guymcswain, 2020-03-03:
Thanks for pointing that out.  I was just observing  how the `wave2Cbs()` function was building the CBs and OOLs and noticed flags were in the mix.

The path I'm pursuing, if it works, should be compatible with flags.

guymcswain, 2020-03-03:
@joan2937 , How do I compile without the built-in tests (x_pigpio.c) executing?  I have deliberately broken the `gpioWaveCreate()` api but just want it to compile anyway.

guymcswain, 2020-03-03:
I see now that it is the tests themselves that are failing to compile.

guymcswain, 2020-03-03:
First problem:
```C
#define CBS_PER_OPAGE 118
#define OOL_PER_OPAGE  79
```
The total CBs and OOLs are scaled by the above constants.  To greatly simplify things, I need to force these constants to be the equal.  Let me if this is not possible and what relationship needs to be maintained between these two quantities.

guymcswain, 2020-03-04:
@Jul3k , I have pushed an experimental branch, **'wavesize'**, which allows the creation of fixed size waves.  I modified the script from the top of this thread to create 4 fixed size waves using:
```python
new_id = pi.wave_create(4)
```
The script now runs to completion.  I did not hook up my scope to observe the waveforms, yet.

Give it a try and let me know how your application responds.

guymcswain, 2020-03-04:
Latest commit 58c5ad756bb751fab80619ac89d19fbe8bbd8829

Errata:
* size parameter must be greater than 2
* flags are not supported

Jul3k, 2020-03-04:
@guymcswain, great job! Thank you! It seems to run perfectly after some testing. I can confirm that size must be greater than 2. Just a note from my side: _The name size might be misleading as it does not really reflect the size. Maybe something like `splits` or `max_waves` would be better?_

guymcswain, 2020-03-04:
Can you observe your generated waveforms?  I haven't put my test together to do so.  I worry that the 'dummy' cb that got added at the end of the wave might have side effects.

guymcswain, 2020-03-04:
@joan2937 ,

> The READ flag is only used by rawWaveAddSPI.

Consistent with the above usage, can we generalize and say the usage of *flags* and *ools* are mutually exclusive within a given wave?

> The TICK flag is only used by some example code (pps.c) which generates a pulse on the second.

I could not find the file `pps.c`.  I assume it also uses flags exclusive of *ools*.

joan2937, 2020-03-04:
@guymcswain 

I wouldn't make any assumptions about flags and ools being mutually exclusive.

I can only comment on my usage of the library.  I'd be loathe to remove anything someone else might be relying on.

guymcswain, 2020-03-04:
I agree not to remove/break anything that users could be relying on.  That said, users shouldn't rely on anything that is undocumented.  I'm having difficulty finding anything in the documentation that explains the usage of flags with waves.  If no such documentation exists, then it should be fair game to change but then nail it down with documentation.  Its good to think of the documentation of the APIs as a contract.

guymcswain, 2020-03-04:
Forgive my rambling, but, I do think there are valid use cases for flags in waves.  To my understanding, flags in waves allow you to create a sort of "input wave" where you can get time stamped level information similar to what the `alertEmit()` thread does for notifications.  Mutual exclusion would still allow these cases.  

What puzzles me is a use case for a simultaneous output (ools) and input (flags) waveform.  Also, it appears there may be a conflict with both waves using a common OOL buffer.

guymcswain, 2020-03-04:
@Jul3k , which part of that comment are you reacting to?

Jul3k, 2020-03-04:
@guymcswain, I didn't react to your comment. I tested your addition today running 4 steppers simultaneously with microstepping and it worked perfectly. I did not see or hear any jerk. I also looked at the timings with a logic analyser and saw no jitter. Step frequencies up to 100kHz for infinite time 😃

Jul3k, 2020-03-04:
@guymcswain, yes probably. I'm gonna habe a look at it tomorrow.

guymcswain, 2020-03-04:
New function is possible but is a lot of work to change all the client's code and documentation.

Jul3k, 2020-03-05:
@guymcswain, it is not possible to have variadic functions also accepting no parameters, see this [Question on Stackoverflow](is-it-possible-to-create-a-c-varargs-function-with-no-arguments). I would go for a new function `wave_set_resource_splits(int n)` or having a dedicated `wave_create` function. What is your oppinion?

joan2937, 2020-03-05:
Perhaps

```
gpioWaveCreatePad(int percent) C 
wave_create_and_pad(percent)   Python
wave_create_and_pad(percent)   pigpiod_if2
WVCAP percent                  pigs
```
Where the percent gives the percentage of the resources to use (in terms of the theoretical maximum, not the current amount free).

So if you wanted to have three waves being rotated you could pad each to 25% leaving another 25% free for any other waves which might be wanted.

Jul3k, 2020-03-05:
Should it be percent or fraction (total number of waves)? Percent has the problem that e.g. to have 3 waves would require 33.3333 percent set or mean to loose some recources.

joan2937, 2020-03-05:
I would be happier with a percent. In practice I reckon we are only talking about a few waves.  The percent could be specified out of a million as done for hardware PWM dutycycle.

There are going to be wasted resources whatever happens with padding.  It is not always trivial to create a wave which fills a particular hole (if it was there would be no need for this function).

guymcswain, 2020-03-05:
Ok, thanks for direction on APIs.  Do we have agreement on this statement:

> "Standard waves and padded waves can coexist, but padded waves do not support flags."

Jul3k, 2020-03-05:
Would flags in principal allow to create a callback once a certain pulse has been transmitted by the DMA? There might be use for a callback once a certain position has been reached in the example of stepper motor control.

guymcswain, 2020-03-05:
From what I can see in the `rawWaveAddSpi()` or `rawWaveAddGeneric()`, flags allow a wave to be constructed that collects system tick or input levels (while simultaneously outputting levels?).  I don't think it gives us insight into the position of the transmission.

Jul3k, 2020-03-05:
I added the functions as @joan2937 recommended and included your changes @guymcswain https://github.com/Jul3k/pigpio/commits/wavesize

joan2937, 2020-03-05:
> Ok, thanks for direction on APIs. Do we have agreement on this statement:
> 
> > "Standard waves and padded waves can coexist, but padded waves do not support flags."

That's fine by me.  If you think they could be added in the future I would just say "do not currently support flags".  Otherwise it's fine as is.

guymcswain, 2020-03-05:
@Jul3k , Thanks I'll take a look at that shortly.

@joan2937 , I need your thoughts on the handling of flags.  The code snip below, I believe, shows the crux of the problem with flags on padded waves:
https://github.com/joan2937/pigpio/blob/58c5ad756bb751fab80619ac89d19fbe8bbd8829/pigpio.c#L9619-L9628

The middle condition constrains the shared memory pool of OOLs to be non-overlapping between the allocation for numBOOL (OOLs that DMA writes to gpio registers) and numTOOL (OOLs that DMA reads from the gpio register or system timer).  

I initially tried to allocate half OOL resource to BOOL and half to TOOL but the OOL resources were consumed too quickly.  Meaning it failed for waves padded to 50% (and possibly down to 25%, but I can't remember).   Nevertheless, it misses the 'sweet spot' for padded wave applications in my opinion.  So in the current experiment I just allocated everything to BOOL and nothing to TOOL.  This is was I mean by flags are not supported.

Just because I can't imaging a use case for flags in the case of generating a continuous output waves doesn't mean we shouldn't support flags in padded waves - I've been wrong before.  So should we decide to support flags in padded waves we would need to add arguments to specify the proportion of CBs, BOOLs and TOOLs in the new APIs.

guymcswain, 2020-03-05:
@Jul3k , nice!  To enable both of us to collaborate on the same branch, you can open a PR from your fork against the 'develop' branch in pigpio.  That way we both should be able to pull/push as needed until we are ready to merge.

I'm out the remainder of today.  My thoughts are that we should look to make the APIs capable of accepting the additional parameters in the future - (%[, %, %]).

[2 more replies]

closed 2020-04-30 by guymcswain