joan2937/pigpio #127: When is the first alert callback called?

fivdi, 2017-05-01:
I'm currently implementing code in JavaScript to measure proximity with a HC-SR04 ultrasonic ranging sensor using the [pigpio](https://github.com/fivdi/pigpio) Node.js package.

It's going well, but there appears to be a timing issue. It looks like it's necessary to wait for a certain amount of time before alert callbacks function correctly. 

It also looks like the problem can be reproduced in C. For example, the [Sonar Ranger Example](http://abyz.co.uk/rpi/pigpio/ex_sonar_ranger.html) functions correctly. However, the example uses the following line of code to generate a trigger pulse every 50ms:
```c
   gpioSetTimerFunc(0, 50, sonarTrigger); /* every 50ms */
```
If I'm not mistaken this line of code will result in a delay of 50ms before the first trigger pulse is generated. It looks like the echo pulse corresponding to the first trigger pulse will not be detected without this delay.

Below is a slightly modified version of the Sonar Ranger Example. It only generates a single trigger pulse and there is no 50ms delay before the trigger pulse is generated. I would expect this modified version of the Sonar Ranger Example to detect a single echo pulse but it doesn't.

Should this modified version of the Sonar Ranger Example detect a single echo pulse?

```c
#include <stdio.h>

#include <pigpio.h>

/*

P1  Name  gpio    used for

 2  5V    ---     5V
 6  GND   ---     Ground
24  CE0   8       Sonar echo
26  CE1   7       Sonar trigger

*/

#define SONAR_TRIGGER 23
#define SONAR_ECHO    24

/* forward prototypes */

void sonarTrigger(void);

void sonarEcho(int gpio, int level, uint32_t tick);

int main(int argc, char *argv[])
{
   if (gpioInitialise()<0) return 1;

   gpioSetMode(SONAR_TRIGGER, PI_OUTPUT);
   gpioWrite  (SONAR_TRIGGER, PI_OFF);

   gpioSetMode(SONAR_ECHO,    PI_INPUT);

   /* monitor sonar echos */

   gpioSetAlertFunc(SONAR_ECHO, sonarEcho);

   /* generate a single trigger pulse */

   sonarTrigger();

   while (1) sleep(1);

   gpioTerminate();

   return 0;
}

void sonarTrigger(void)
{
   /* trigger a sonar reading */

   gpioWrite(SONAR_TRIGGER, PI_ON);

   gpioDelay(10); /* 10us trigger pulse */

   gpioWrite(SONAR_TRIGGER, PI_OFF);
}

void sonarEcho(int gpio, int level, uint32_t tick)
{
   static uint32_t startTick, firstTick=0;

   int diffTick;

   if (!firstTick) firstTick = tick;

   if (level == PI_ON)
   {
      startTick = tick;
   }
   else if (level == PI_OFF)
   {
      diffTick = tick - startTick;

      printf("%u %u\n", tick-firstTick, diffTick);
   }
}
```

joan2937, 2017-05-01:
I've only had a quick scan through the question.  Perhaps the firstTick logic in sonarEcho() is the issue, perhaps a one-off system will need to be initialised in a different way?

fivdi, 2017-05-01:
>Perhaps the firstTick logic in sonarEcho() is the issue, perhaps a one-off system will need to be initialised in a different way?

I don't think this is the issue. sonarEcho() never gets called in the above code. This can be seen by adding `printf("sonarEcho\n");` as the first line of sonarEcho(). The text "sonarEcho" never gets printed to the console indicating that sonarEcho() is not called.

joan2937, 2017-05-02:
Yes, you are right.  I think it's because gpioInitialise() returns once all the processing threads have been started.

I'll need to check the detail but I think the root cause is that the alert thread needs one or two complete "DMA cycles" before it starts generating alerts.  By "DMA cycle" I mean the filling of the cyclic buffer of read DMA GPIO events.

If you add a sleep of a tenth of a second immediately after gpioInitialise() you will probably capture the first alert.

This isn't an area I am fond of touching because of the risk of breakage.

I'll see if the delay is necessary or was just something I thought of as a "good idea" at the time.

fivdi, 2017-05-02:
I can confirm that adding a delay after calling gpioInitialize() fixes the problem on the JavaScript side (https://github.com/fivdi/pi-io/blob/master/lib/pi-io.js#L17-L25). It's a 50ms delay at the moment but it can be increased to 100ms. That being said, if it's possible to fix the problem by adding a 100ms delay after calling gpioInitialize(), would it not be better if gpioInitialize() added a 100ms delay before returning? Then everyone using alerts would benefit from the change.

fivdi, 2017-05-02:
If you would prefer not to modify pigpio it's ok for me to deley for 100ms in my code.

joan2937, 2017-05-02:
I suggest you do that for the time being.  I need to understand what's going on, my memory is not consistent with the current code.

joan2937, 2017-05-11:
Okay I have had a better look at the code and have re-jigged initialisation slightly.

This means the alert thread will be operational before gpioInitialise() returns.

This increases the default average initialisation time marginally from 0.320 to 0.328 seconds.

I'll make the change in V63 which will be released in the next day or so.

fivdi, 2017-05-11:
Excellent. Thanks a million.

joan2937, 2017-05-13:
V63 has just been released.

fivdi, 2017-05-14:
:thumbsup: Great Job

I just gave V63 a try and everything functioned as expected.

closed 2017-05-14 by fivdi