joan2937/pigpio #128: Is it possible to get input events when using pigpiod?

mattjlewis, 2017-05-05:
Hi Joan, is it possible to request to receive asynchronous input events when using pigpiod? I've just implemented support for remote socket communication in my pigpioj Java library but this is the only area I couldn't see how to implement without polling.

Regards,
Matt

joan2937, 2017-05-05:
What sort of input events are you thinking of?

pigpio is pretty much a GPIO library, and GPIO events are handled by callbacks.

mattjlewis, 2017-05-05:
Sorry I should have been clearer. I was wondering if it was possible to get GPIO input events when using pigpiod. 

Regards,
Matt

joan2937, 2017-05-05:
That's no problem.

You would need to callbacks in the way the Python module and the pigpiod_if2 library do.

They dedicate a socket to receive socket notifications which are (local) pipe notifications redirected to a socket.

Have a look at callback handling in https://github.com/joan2937/pigpio/blob/master/pigpiod_if.c or https://github.com/joan2937/pigpio/blob/master/pigpio.py

mattjlewis, 2017-05-05:
Great, thanks I'll take a look now. Didn't spot that in the socket interface documentation: http://abyz.co.uk/rpi/pigpio/sif.html.

closed 2017-05-05 by mattjlewis

joan2937, 2017-05-05:
Yes, it should be covered somewhere on that page.  When I get time I'll add a description, don't hold your breath though.