joan2937/pigpio #129: Please clarify the difference between pigpio and pigpiod C interface

bschousek, 2017-05-09:
In what cases would a person choose to use the pigpio C interface instead of the pigpiod C interface? Are there performance differences? It appears a similar if not identical set of functions are included.

joan2937, 2017-05-10:
You pays your money and you takes your choice.

The proper choice depends on what you plan to achieve.

A simplistic assessment is

1. C linked against pigpio. Fastest code, slowest development. Only one program linked against the pigpio library can be running at a time (the program in effect becomes the pigpio daemon).  Program can only run on the Raspberry Pi.

1. C linked against pigpiod_if2 (via pigpio daemon): Fast code, fast development.  Requires the pigpio daemon to be running.  Many programs can be running at any one time talking to the pigpio daemon.  Programs can be compiled and run on any machine with a suitable compiler, e.g. Linux, Windows, Mac.

1. Python (via pigpio daemon). Slowest code, fastest development.  Requires the pigpio daemon to be running.  Many programs can be running at any one time talking to the pigpio daemon.  Programs can be run on any machine with a suitable Python interpreter, e.g. Linux, Windows, Mac.

The accuracy of PWM,  Servo pulses, callbacks etc. is the same regardless of the above choice.

SlySven, 2017-05-10:
Choice 1. is what you need MHO if you have to protect certain pins / functions again spurious changes from proto-type or unprivileged code - I have a UPS daemon running on one of my RPis and that needs exclusive control/access: to a GPIO pin used to monitor a critical function (power WILL be GONE 45 seconds after pin is taken low by UPS ⚡️ ); and to: the serial port which it uses for frequently and regular communicating with the UPS to monitor other status details.

In that situation where this daemon is started in `/etc/rc.local` {I'm a sysV `init` fan who thinks `systemd` is daemon spawn rather than it being something to spawn daemon :rofl:} with root privileges it is not unreasonable for it to provide the functions of the `pigpiod` daemon - though as I mentioned in #63 it meant I had to tweak the signal handling otherwise it would die as soon as it got signals associated with debugging it!

joan2937, 2017-05-10:
Yes @SlySven, good point.

With Choice 1 you get much more control of who is allowed to manipulate the GPIO and how.

The permissions to access GPIO may be set (read only or read write) as well as enabling/disabling the socket and pipe interfaces.

SlySven, 2017-05-11:
Though if you do go that route you'd be advised to have something to check for the PID file the daemon or other program acting as it before trying to instantiate another daemon providing the library functions - this is what I'm currently using:

```
    // ... near top of main() ...
    // Test for existance of lockfile as proof of another instance of this or pigpiod
    // already running
    int lockFilePid = initCheckLockFile();
    if( lockFilePid > 0 ) {
        syslog( LOG_WARNING, "an instance of this, the pigpiod daemon or some other program using the pigpio library is already running (as Process %i) and only one may operate at any time please investigate that process and kill it before trying to run this program again.",
                lockFilePid );
        closelog();
        exit( EXIT_FAILURE );
    }
    else if( lockFilePid == -1 ) {
        syslog( LOG_WARNING, "an instance of this, the pigpiod daemon or some other program using the pigpio library is already running and only one may operate at any time please investigate that process and kill it before trying to run this program again." );
        closelog();
        exit( EXIT_FAILURE );
    }
    // ...
```

The define for `PI_LOCKFILE` is located in the `pigpio.h` that you will of course need to `#include` :

```
// Returns:
// (positive) PID of instance of daemon already running!
// 0 if there is no lockfile
// -1 if there is a lockfile but cannot make sense of contents
int initCheckLockFile(void)
{
    int fd;
    int count;
    int pid = 0;
    char str[20];

    fd = open( PI_LOCKFILE, O_RDONLY );
    if( fd != -1 ) {
        pid = -1;
        count = read(fd, str, sizeof(str)-1);
        if( count ) {
            pid = atoi( str );
        }
        close( fd );
    }
    return pid;
}
```

bschousek, 2017-05-16:
Thanks for all the thoughtful answers.

closed 2017-05-16 by bschousek