joan2937/pigpio #130: Question: Problem after transmitting a signal

TheRaspyDev, 2017-05-12:
I'm using your great library to receive and transmit 433MHz codes (in python). I want to replace my "hardware" remote control with this.
Everything works now. But there's one thing I don't understand:

When I send a code/signal, the output pin seems to continue sending a signal. I call pi.wave_clear() after transmitting the signal.
My real hardware remote control doesn't work anymore after transmitting a signal with pigpio.
I found a workaroud: When I add pi.set_mode(self.gpio, pigpio.INPUT) at the end of my python script, my hardware remote control works again.
Any idea on that?

Regards, Maurus

joan2937, 2017-05-12:
You need to prepare a hopefully small example script which demonstrates the problem.  I will then be able to run the script and confirm or deny the problem.

TheRaspyDev, 2017-05-12:
Thanks for the fast reply. Here's the script (an adapted copy of your _433.py) and a json file containing the codes.
To run the script, type "python _433.py tx -f "test.json" -c a_on".
When I comment the line #302 (pi.set_mode(self.gpio, pigpio.INPUT)) my remote control doesn't work after running the script.
When I include the line #302 (as in the example), my remote control continues to work after running the script.

[433_Analyze.zip](https://github.com/joan2937/pigpio/files/997331/433_Analyze.zip)

joan2937, 2017-05-12:
Before I look in detail could I just check it's not simply because the transmitter is left enabled.  What happens if you change #302 to **pi.write(self.gpio, 0)**?

TheRaspyDev, 2017-05-19:
Thanks for the hint, it works :)
But I still don't understand why I have to do this explicitely when I delete all waves with **pi.wave_clear()** before.

electrokean, 2017-05-20:
It is because you are potentially leaving the gpio pin in a logic 1 state, and thus leaving the transmitter active.  Deleting the waves won't change the state of the gpio, and just leave it in its last state.
You could add a final low going pulse at the end of your waveform if you end the loop with current_level as false (i.e. when the previous pulse when high).  You'd do this just before line 256.
It is probably easier to just include to the line suggested by @joan2937 above to always force it low.

joan2937, 2018-02-09:
This seems to have been resolved by ensuring the transmitter is off once the signal has been sent.

If you think I need to update any pigpio examples let me know.

closed 2018-02-09 by joan2937