joan2937/pigpio #131: More efficiently reading the socket from python

matthijskooijman, 2017-05-17:
While debugging some performance problems (at around 4000 notifications per second, my rpi could not keep up and the socket buffer would fill up causing latency), I made some changes to the python code to read more efficiently from the socket. The idea is to call `recv` to read more than one packet at a time, which should reduce the number of syscalls (not verified) and thus be more efficient. In practice, this didn't work well enough to solve my problem completely, but I wanted to share the change anyway.

I'm probably switching to a different setup, so I won't be needing this code anymore, but I wanted to share it anyway in case it's useful (but I don't want to spend the time on a proper pullrequest, so here's the patch inline):

```
diff --git a/pigpio.py b/pigpio.py
index df01c0d..371115f 100644
--- a/pigpio.py
+++ b/pigpio.py
@@ -1110,17 +1110,19 @@ class _callback_thread(threading.Thread):

       lastLevel = self.lastLevel

+      RECV_SIZ = 4096
       MSG_SIZ = 12

+      buf = bytes()
       while self.go:

-         buf = self.sl.s.recv(MSG_SIZ)
+         buf += self.sl.s.recv(RECV_SIZ)
+         offset = 0

-         while self.go and len(buf) < MSG_SIZ:
-            buf += self.sl.s.recv(MSG_SIZ-len(buf))
-
-         if self.go:
-            seq, flags, tick, level = (struct.unpack('HHII', buf))
+         while self.go and len(buf) - offset > MSG_SIZ:
+            msgbuf = buf[offset:offset + MSG_SIZ]
+            offset += MSG_SIZ
+            seq, flags, tick, level = (struct.unpack('HHII', msgbuf))

             if flags == 0:
                changed = level ^ lastLevel
@@ -1143,6 +1145,7 @@ class _callback_thread(threading.Thread):
                   for cb in self.events:
                      if cb.event == event:
                         cb.func(event, tick)
+         buf = buf[offset:]

       self.sl.s.close()

```

joan2937, 2017-05-17:
Thanks for that.

I batch reads in piscope, pigpiod_if.c, and pigpiod_if2.c but may have assumed it wouldn't make a significant difference in Python.

Did you get a measure of the performance improvement?

matthijskooijman, 2017-05-17:
> Did you get a measure of the performance improvement?

It seemed a little better, but I haven't done thorough measurements (with our without this patch, python was too slow to handle 4k events per second, I haven't tried throttling the events to find the turnover point). Since I'll be using a different setup, I won't be able to invest more time in investigating this, so feel free to measure yourself, or ignore my patch if that's better :-)

joan2937, 2017-05-17:
I have some tests on a original PiB and a Pi3.

There does seem to be a good performance increase for Python running on the Pi but no benefit for Python running over the network (i.e. script running on a remote PC talking to a Pi).

The following figures are the events per second handled (an event in this case is a callback call).

```
              unpatched patched improvement
Pi B (local)  10k       15k     50%
Pi 3 (local)  67k       80k     19%

Pi B (remote) 89k       89k     0%
Pi 3 (remote) 81k       81k     0%
```

christianeisendle, 2017-09-22:
I'm currently working on C#/.Net bindings for pigpiod and while integrating this batch-reading optimization, I discovered a bug in the diff above. 

```
+  while self.go and len(buf) - offset > MSG_SIZ:
                                      ^^^^
```
`len(buf)` should be greater than or equal to `MSG_SIZ` otherwise the last/only message in the buffer is ignored.

matthijskooijman, 2017-09-23:
Good point, that should indeed be >=.

joan2937, 2018-02-08:
This has been added to V65.

closed 2018-02-08 by joan2937