joan2937/pigpio #133: Timer function is no more working after time setting in the future

agsrpi, 2017-05-24:
I made this little piece of software:
```
void TimerSendCheck()
{
  static int counter=0;
  if(++counter>100) {
    printf("100 check interrupts\n");
    counter=0;
  }
}

void InitTimer()
{
    gpioCfgInterfaces(PI_DISABLE_FIFO_IF | PI_DISABLE_SOCK_IF);
    if(gpioInitialise() == PI_INIT_FAILED)
    {
        printf("gpioInitialise failed!\n");    
    }
    else
        printf("gpioInitialise o.k.\n");

    gpioSetTimerFunc(1, 50, TimerSendCheck );
}
```
Everything seems to work, until I set a date in the future: the program stops printing the message each 100 interrupts.

The timer does not restart also if I resume the last time or I put a time in the past:
```
root@SLP-demo003:~# tail -f /var/log/syslog
Jan  1 00:29:17 SLP-demo003 timer_test[30967]: 100 check interrupts
Jan  1 00:29:19 SLP-demo003 timer_test[30967]: 100 check interrupts
Jan  1 00:29:21 SLP-demo003 timer_test[30967]: 100 check interrupts
Jan  1 00:29:23 SLP-demo003 timer_test[30967]: 100 check interrupts
Jan  1 00:29:25 SLP-demo003 timer_test[30967]: 100 check interrupts
Jan  1 00:29:27 SLP-demo003 timer_test[30967]: 100 check interrupts
Jan  1 00:29:29 SLP-demo003 timer_test[30967]: 100 check interrupts

root@SLP-demo003:~# date 12300100
Sat Dec 30 01:00:00 UTC 2017
root@SLP-demo003:~# date 01010000
Sun Jan  1 00:00:00 UTC 2017
root@SLP-demo003:~# tail -f /var/log/syslog
    (no message from timer_test)
```

Did you ever notice such behaviour?

joan2937, 2017-05-25:
No, I have not noticed such behaviour.

The clock ticks will restart printing when the next tick is due in absolute time.  An absolute time clock is used.  It may be better to use a relative clock for this function, however I have no way of knowing if doing so will break others existing code.

Why do you need such clock discontinuities?

agsrpi, 2017-06-26:
Hello Joan,

sorry I am replying so late, but I got involved into another project and forgot to check the emails from github.

Anyway, my problem arises as I need a clock reference for the raspberry to do timed tasks (in crontab).

Usually, at the start, fakehwclock gives an estimate of the time, but then the NTP corrects the time when the network is on.

What is happening is that, often, the raspberry remains in off state for days and, sometimes, as I turn it off without a proper shutdown, the fakehwclock file is not updated so it gives a date far in the past.

Once I updated the clock with an NTP I set up on my PC, but due to time zone misalignment it set a time off by 12 hours. Anyway, when I corrected the time the timer did not start.

Let me know if you plan to change this behaviour or to add a different api for the "relative" ticks.

joan2937, 2017-06-26:
I think the way you want it to work is the "correct" way and what I would probably do if starting from scratch.

I am minded to change the behaviour.

However I am taking a "holiday" from pigpio at the moment so don't expect any changes for a month or so.

agsrpi, 2017-06-26:
OK, I'lll "stay tuned".

Thank you.

joan2937, 2018-02-13:
This will be fixed in V66 (by simplifying timer loop).

closed 2018-02-20 by joan2937