joan2937/pigpio #134: Initialize is spelled incorrectly.

martindeegan, 2017-06-07:
Not a big problem, but it messed me up for a long time when trying to make a wrapper.

[](http://abyz.co.uk/rpi/pigpio/cif.html#gpioInitialise)

joan2937, 2017-06-07:
In English initialise and initialize are equally valid spellings.  I'm not going to change the function name.

closed 2017-06-07 by joan2937

fivdi, 2017-07-08:
An alternative non-breaking change would be to introduce a new function called gpioInitialize that simply calls gpioInitialise.

> In English initialise and initialize are equally valid spellings.

I don't think this is 100% correct. In British English initialise and initialize are equally valid spellings but in American English this is not the case. As far as I know the rule is that verbs that can be spelled with either -ise or -ize at the end in British English are always spelled with -ize at the end in American English.