joan2937/pigpio #136: ImportError: No module named 'vw'

Wanjiaa, 2017-06-12:
I have succesfully installed pigpio but after trying to run the example code for VIRTUAL_WIRE (vw.py) for RF, I get `ImportError: No module named 'vw'`.

joan2937, 2017-06-12:
I assume you have altered the file as line 331 does not contain an import.

Wanjiaa, 2017-06-12:
EXAMPLES/Python/VIRTUAL_WIRE/vw.py line 331: `import vw`

joan2937, 2017-06-12:
I'd forgotten they existed.  I use http://abyz.co.uk/rpi/pigpio/examples.html#Python_vw

Either way the file has been altered as neither file gives an import error on my system.

Wanjiaa, 2017-06-12:
No more import errors, but even though the `tx.put()` returns a `true` nothing is received by the `rx.get()`. Range is not the problem because they are next to eachother, rx is on `GPIO 20` and tx on `GPIO 21` as written in the code, pigpiod is also running. How can I get the receiver to receive the "Hello World" the transmitter is sending out? I have tried the receiver with a working versie for arduino (tested it arduino -> arduino) but also nothing.
(My goal is to send data from my arduino and receive it with my raspberry)

joan2937, 2017-06-12:
Firstly check that you are using the correct GPIO.  GPIO 20/21 are connected to expansion header pins 38/40.  Consider using [piscope](http://abyz.co.uk/rpi/pigpio/piscope.html) so you can see what is being transmitted/received.  For any sort of distance (more than a centimetre or so) you may need to attach aerials to the transmitter/receiver (a 16-17 cms long piece of wire).

Wanjiaa, 2017-06-12:
According to piscope ([screenshot](http://prntscr.com/fiw0kd)) I have high and lows both on GPIO 20 & 21, transmitter and receiver are almost touching each other. But when running vw.py I don't get any prints ([screenshot](http://prntscr.com/fiw18h))

Wanjiaa, 2017-06-13:
After 5 minutes the code stops automatically (which is suppose to happen) and returns some numbers but I can't make anything up with these numbers. ([screenshot](http://prntscr.com/fj57gm))

joan2937, 2017-06-13:
I would not run the script within IDLE.  I have no idea what IDLE may be doing in the background.  Run the script from a command shell.

Wanjiaa, 2017-06-13:
I don't have a lot of experience on this or I could say close to none, I tried looking for an other python interpreter but can't seem to find one. Any idea for one?

joan2937, 2017-06-13:
The interpreter is fine.  IDLE is a GUI/shell around the interpreter, a sort of IDE.

Just open a command shell (there should be a menu option), navigate to the directory containing vw.py and at the prompt enter ./vw.py.

Wanjiaa, 2017-06-13:
Both in pycharm through remote host running the script and in the terminal on the pi itself both are showing piscope high and lows but still nothing is printed out.

Wanjiaa, 2017-06-13:
Would it even be possible to have arduino send data and raspberry pi receive it through RF?

joan2937, 2017-06-13:
Yes, if you look at the comments that is one of the tests I did.

I have just hooked up a transmitter to 21 and a receiver to 20 and the software works as expected.

```
$ ./vw.py
0001
Hello World #0001!
0002
Hello World #0002!
0003
Hello World #0003!
0004
Hello World #0004!
0005
Hello World #0005!
0006
Hello World #0006!
0007
Hello World #0007!
0008
Hello World #0008!
0009
Hello World #0009!
0010
Hello World #0010!
```
I don't know what is wrong with your set-up, but it doesn't appear to be an issue with pigpio or the example.

closed 2017-06-13 by joan2937

roysG, 2020-10-11:
> use http://abyz.co.uk/rpi/pigpio/examples.html#Python_vw

The link is not working any more, how do i need to download this library?