joan2937/pigpio #137: Wave output on Pi 3

matherp, 2017-06-13:
I'm using wave output to generate a serial data stream using gpioWaveAddSerial(). On the Pi Zero W this works perfectly. On the Pi 3 the timings are incorrect. I've emailed Jean about this and it is a known issue with the Pi 3 with no obvious reason or systematic way to correct. 

I've now done some fairly precise measurements and the discrepancy can be corrected with the following:
`if(gpioHardwareRevision()==0xa02082 || gpioHardwareRevision()==0xa22082) baud *= 1.128;

This seems to be robust.

If anyone has any idea what may be the origin of this sort of factor I'd be interested to know. Also, are there any other "Pi's" that need this sort of correction?

joan2937, 2017-06-13:
Closing this as in effect it is a duplicate of #80

closed 2017-06-13 by joan2937

matherp, 2017-06-13:
Joan

 

I’m completely unclear from the referenced threads the current status of this issue hence opening the issue. I’m running headless, no audio and need this factor of 1.128 to get the correct wave output. Is my experience standard? If I distribute code with the frig factor will the code run correctly on other peoples’ Pi 3 or is it in some way hardware/OS dependent?

 

Thanks

 

Peter

 

Closed #137 <https://github.com/joan2937/pigpio/issues/137> .

joan2937, 2017-06-13:
Nobody knows what causes the problem.  I have never managed to duplicate the problem on my Pi3.  I have no idea of the ratio of "properly" functioning Pi systems to "improperly" functioning Pi systems.

matherp, 2017-06-13:
Joan

 

Would you like me to loan you my Pi 3 (including SDcard) so that you can test further? If yes let me have an address. I would need the SDScard back but would happily give up the Pi 3 as a gesture of thanks for all your efforts to support C

 

Rgds

 

Peter

joan2937, 2017-06-13:
You could try adding the entry `core_freq=250` to `/boot/config.txt` and rebooting to see if that resolves the problem.

electrokean, 2017-06-13:
I was also going to suggest setting core_freq to 250
See https://github.com/raspberrypi/linux/issues/2060

matherp, 2017-06-13:
No change – still works fine with my 1.128 fudge factor