joan2937/pigpio #138: Java Native Interface sigHandler: Unhandled signal 11, terminating

LucosidE, 2017-06-19:
When using pigpio in Java through JNI the program terminates with

sigHandler: Unhandled signal 11, terminating

this happens when using interrupt functions such as gpioSetAlertFunc or gpioSetISTFunc even if these functions do absolutely nothing.

This might be a JNI issue. If anyone has any ideas on how to use pigpio interrupts in Java I would greatly appreciate any suggestions.

I updated to latest java
java version "1.8.0_131"

```
void interrupt_s(int gpio, int level, uint32_t tick)
{
   /*
      Callback does nothing
   */
}

JNIEXPORT jlong JNICALL Java_pigpioTest
  (JNIEnv *env, jobject obj, jint pin)
{
	//Initialize PIGPIO library
	if (gpioInitialise() < 0) {
		return 0;
	}
        // Set a single pin as input
	gpioSetMode(pin, PI_INPUT);
	gpioSetPullUpDown(pin, PI_PUD_DOWN);
	   /* monitor pin level changes */
	gpioSetAlertFunc(pin, interrupt_s);
	return 1;
}
```

The above runs for a while and then the sigHandler: Unhandled signal 11, terminating error happens.

SlySven, 2017-06-22:
Signal 11 is "Segment Violation" IIRC which normally is a __fatal__ one.  Means that the code was using a pointer that wasn't pointing at something valid (or it was pointing to the "null" value! 😮 ) - and it is not something that can be "Handled" - you just need to make sure it doesn't happen.  OTOH the message "sigHandler: Unhandled signal 11, terminating" sounds like it is the `pigpio` library itself that is dying - it is started by the `pigpiod` or a user provided replacement that uses the `pigpio` library directly. I am not sure if that is what JNI does as I have not heard of that before - but then there is a lot I don't know! 😀

afterhoursengineering, 2017-08-02:
I am using JNA instead of JNI with the libpigpio.so library and I get the 
`sigHandler: Unhandled signal 2, terminating` 
and the 
`sigHandler: Unhandled signal 4, terminating`
...and I think I've gotten the "signal 11" one also...
So far, on the Java-side, the only pigpio function I'm calling is `gpioVersion()`...

I've seen some discussion of this and in one of them, I believe Joan concludes this may be an issue with JNI/JNA.  She recommends using pigpio daemon through sockets rather than directly through Java.  
https://www.raspberrypi.org/forums/viewtopic.php?f=81&t=77337
https://github.com/fivdi/pigpio/issues/6

I have begun testing this theory using some code from https://github.com/nkolban/jpigpio, specifically the PigpioSocket.java portion (and associated dependencies) and I've been sending and receiving I2C data reliably so far without the unhandled signal crashes.

If you go this route, don't forget you have to start the pigpiod daemon first. `sudo pigpiod`

jw3, 2017-08-02:
Ive been battling on and off in hobby time with jni and pigpio for over two years.  Here are some notes from the last time https://github.com/jw3/test-case-pigpio-jvm-crash

My primary testing was an arduino that just blasts events to the pi.  Sometimes it runs for days, sometimes seconds, but always a sig 11.  I was trying to develop a more reliable testing capability before walking away the last time.

Ive used jni before in other projects without issue, but not on the pi or arm.   Someone with more experience debugging low level jvm crashes might have luck identifying the problem.

There are some other repos of mine that went down similar paths, might be useful.

Its worth noting that there are at least a couple of jni pigpio libraries that appear to be somewhat mature.  Have always meant to grab one and see if they fail in the same way that my attempts do (ie the arduino test).  Never got around to that, always was one hack away from getting it working :cold_sweat:

joan2937, 2017-08-07:
pigpio traps all but a few signals and aborts if they are seen.  It does this to ensure the DMA engines are safely stopped on program termination.

This does not mean a pigpio error caused or didn't cause the signal, just that it noticed the signal.

If you think pigpio is causing the signal a reproducible test will be useful (hopefully with no hard to build dependencies).

dvlopt, 2017-09-11:
Joan, first of all thanks for your library, it is very complete.

There are 2 ways of calling native code from java: JNI and JNA. JNI requires you write some C glue code for creating a bridge between the jvm and the native code you are calling. JNA on the other hand simply loads native libs and figures out how to call native functions from java classes you've written. It handles type conversion etc. For instance, if i needed only gpioWrite my class would look like this :

```java
public class MyPigioWrapper {

    public static native int gpioWrite( int gpio, int level ) ;
}
```

I've been experimenting with JNA and pigpio and unfortunately got segfaults, lots of them. I've been testing with gpioWrite but really it was an omnipresent problem. Now, it looks like this happens when threading is involved. For instance, sequentially creating 1e6 or 1e7 threads that write to a pin will end up with a segfault after a while. When I run it on the main thread, even after 24h everything is fine. I've tested several ways of writing to a pin from threads, threads within threads, etc, and it always end up segfaulting in this context.

It must be because the GC kicks in after a while and clean-up something that's not supposed to be cleaned up. That's why it looks so random. What actually happens is a mystery to me, JNA is quite a beast to understand (as well as JNI). gpioWrite is a pretty simple function from what I read, there really isn't much to break, no secret bookkeeping involving leaky pointers or whatnot.

The thing is, JNA is very mature. Googling for this kind of errors doesn't provide much, people simply don't seem to experience that kind of behavior. Why does it specifically happen with Pigpio ? Because it works very closely with hardware ? Well, I can't reproduce anything like that with wiringPi so all of this is odd, really odd.

I sure hope to be able to use Pigpio in the future.

christianeisendle, 2017-09-22:
The question is why to use JNI/JNA instead of operating at socket level.
I also wanted to use pigpiod in a managed environment (C#/.NET in my case) and decided to implement the .NET bindings at socket level. IMHO the effort is almost the same than writing wrapper functions to call native code from within managed code. Additionally, it avoids issues like those described above.
Of course, it requires reverse-engineering of the socket interface, but existing python/C bindings are a good reference and actually the interface is straight-forward.

guymcswain, 2017-09-22:
+1 for 'operating at socket level.'  I would add the big benefit of offloading your application from a lowly pi zero that may be running pigpiod.

There are, however, some issues to overcome using the SIF.  Some I'm dealing with are:

- Polling buffers to check if data is available to read.
- Polling busy status of the waveforms.
- Can't re-initialize pigpio

Pigpio provides scripting APIs that can do some low level busy-wait-then-trigger-event that can mitigate having to poll across the network.  But there are pit falls if you go the scripting route as Joan admits there needs some tidying up of that code.  Personally, I'd prefer that APIs supporting reading of data have built-in events or notifications - see issue #148. 

As for a 'reset' (sometimes you just need a hammer), I suppose we could write a custom function until Joan returns from 'pigpio holiday'.  🙏

MaitreDede, 2017-09-27:
Hi,

Which one is faster ? Using libpigpio or the socket interface localhost ? Or an other method ?
I though that libpigpio was the fastest, but there are signal conflicts between libpigpio and other runtimes (java, dotnet core, mono...).
So I think for actual version we have to stick to socket interface.

@christianeisendle 
I have written a pigpio C# lib here https://github.com/MaitreDede/RaspberryPi.Libs/tree/master/RaspberryPi.PiGPIO

Usage is simple. It is a work in progress, but there is a common interface (IPiGPIO) that is implemented for direct libpigpio usage ([PiGpio](https://github.com/MaitreDede/RaspberryPi.Libs/blob/master/RaspberryPi.PiGPIO/PiGpio.cs)), Socket interface client ([PigsClient](https://github.com/MaitreDede/RaspberryPi.Libs/blob/master/RaspberryPi.PiGPIO/PigsClient.cs)) and Pigs commands generator ([PigsGenerator](https://github.com/MaitreDede/RaspberryPi.Libs/blob/master/RaspberryPi.PiGPIO/PigsGenerator.cs)).
Can you give it a try please ? :)

christianeisendle, 2017-09-27:
> Which one is faster ? Using libpigpio or the socket interface localhost ? Or an other method ?

I would say libpigpio. For implementing the socket interface pigpiod runs a socket server on top of libpigpio. So there is the overhead of serialization of API calls transport via the socket (involves multiple system calls) and de-serialization on target client. Using libpigpio the target application just directly calls the APIs.
The socket interface is more intended to implement distributed setups like running the application on a different machine. Of course, directly using libpigpio requires the target application to run with appropriate high execution privileges - something an application using the socket interface wouldn't require.

> I have written a pigpio C# lib here https://github.com/MaitreDede/RaspberryPi.Libs/tree/master/RaspberryPi.PiGPIO

I did my own implementation (https://github.com/christianeisendle/pigpiosharp), but only covering limited feature set (I2C, GPIO, Serial) and do not plan to extend this since i'm not able to implement and run required unit tests due to lack of hardware. I need to check yours, maybe I decide to retire mine to have something more complete :)

MaitreDede, 2017-09-27:
@christianeisendle I am going to use mine and maintain it. I have only implemented what I am using primarily (but I one day will cover all pigpio methods).

IMHO, I think that when pigpio is used as library, it is the responsibility of the dev to call properly 
`gpioTerminate` to keep the pi in an clean stable state.
@joan2937  Do you think it may be possible for the lib to either to totally remove the signal handling, or at least deactivate it so it does not conflict with java, c# or other "vm" ?
Also is it possible (for the lib) that instead of logging to console (fprintf) you can let the dev assign logging callbacks ?
Thanks

guymcswain, 2017-09-27:
> Which one is faster ? Using libpigpio or the socket interface localhost ? Or an other method ?

I suggest it doesn't matter as pigpio has already taken care of "speed" for you.  Whether you access the lib directly or via Socket IF the internals use DMA hardware to read/write and timestamp GPIO events - thereby decoupling timing from the calling application.

So back to @christianeisendle 's point, make your life easier and avoid signal handler issues by using the socket.  @MaitreDede , I see you have `PigsClient`.   Embrace it!

MaitreDede, 2017-09-27:
@guymcswain When I talk about speed, I mean the time it takes for a command emmited from my code to be sent and its response received.
For example, I am driving a led matrix display. It can only light 1/4 of the leds, son I have to display all 4 quarters very fast to see a complete display.
When I use `libpigpio`, the display looks fine. No flickering, display is perfect, but I hit the signals issue.
When I use the socket interface remotely (same wired network), we can see which quarter is lit.
When I use the socket locally, the image is flickering.
The socket interface, even used locally, adds too much overhead, like @christianeisendle said. So for a time-critical application, using the socket is not an option.
Like I said, I think the best option is when a dev use pigpio as lib, it is the dev's responsibility to call properly the terminate method, not the lib.

guymcswain, 2017-09-27:
@MaitreDede , Not to dissuade you from using `libpigpio` but the power in pigpio lies within its APIs that leverage hardware to control timing.  If your application is controlling the timing then you are subject to other uncontrollable latencies such as the OS pre-emption.  Although I have not implemented an led matrix driver, I would look to see if the waveform APIs could be employed to provide very precise timing.  (I'll say no more.  Just trying to point out some of the wonderful capabilities of this library : )

MaitreDede, 2017-09-27:
@guymcswain In the case of led panels, they are commonly driven by bit-banging data with a clock signal, like SPI, and some other pins. So precision is not needed. It only needs to be fast enough so humans can see 1 full image instead of flickering :)
The simplest led panel I have seen is the [Freetronics DMD](https://www.freetronics.com.au/collections/all-products/dmd). Pins are Data, Clock, OutputEnabled, A, B, Strobe. Pixel data is sent by bit-banging to Data/clock, then you set the row address you want to display (A,B), and you "strobe" to really display the row. You transmit 1/4 panel for each strobe. So 4 is needed to transmit a full panel, but only one quarter is displayed at a time.
Yes, you are right that I may see flickering when the pi gets loaded, but it will be because the app does not send commands fast enough. The waveform api could be one solution, but this will lead to (i think) big waveforms, that will change a lot (example : scrolling text).

MaitreDede, 2017-09-27:
I have found the pull request https://github.com/joan2937/pigpio/pull/8 that seams to be one solution. In a short term, I can distribute `libpigpio.so` with my lib, but for a long-term solution that avoids me to handle myself pigpio version, what could be the best way ?

guymcswain, 2017-09-28:
@MaitreDede  excellent find!
@LucosidE , @dvlopt , @jw3 and @afterhoursengineering , does the compiler flag option referenced in  #8, resolve your issue?

afterhoursengineering, 2017-09-28:
This looks interesting.  I may try it and will let you know.  Since my original problems with using the lib via JNA, I've done all my development using the socket interface.  For those interested in timing, I'm running on a Raspi3, app in Java, pigpio via sockets.  My quickie speed test was to send 1 byte and read 2 to a slave device as fast as possible.  Average period between cycles is about 850us.  If the compiler option mentioned will make pigpio via JNA stable, I'll conduct a similar test and report out.  Thanks y'all.  I've appreciated the info in this thread.

MaitreDede, 2017-09-28:
I have compiled the lib with the define : `make -j2 CFLAGS+=-DEMBEDDED_IN_VM`, no more errors.
I think @afterhoursengineering will see the same thing.

@guymcswain Do you think that it may be possible in a future pigpio version to add a function `gpioCfgSetSignalHandling(bool enabled)` to disable signal handling ? true by default, to keep existing behavior...

guymcswain, 2017-09-28:
@MaitreDede  Mind you that I'm not the shop keeper - just a customer waiting in queue... 
Should this be considered a documentation issue and we write a PR to update the `readme.md`?  If an API change then `gpioCfgSetInternals()` may be a good candidate.  `gpioCustomx()` are possibilities but I've struggled with how to make these available to other library users.
In any case, until Joan returns we need an ad-hoc approach that allows users to contribute and share.  I'm not GitHub savvy enough to suggest one.

MaitreDede, 2017-09-28:
@guymcswain Ok :)
I have made some code, you can check PR https://github.com/joan2937/pigpio/pull/164. Signal handling can be disabled before initializing the lib.

LucosidE, 2017-09-28:
You guys are fantastic, I've been launching a separate process from Java
and communicating with the process through iostream to avoid the Signal 11
issue with JNI. Excited to give JNI another try.

SlySven, 2017-10-02:
Actually I found there were one or two issues with the signal handling in the library as it was a while back and I hacked together a revision that made the signal handling a bit nicer as a rejected PR #58 - that had some rough edges and I've been away from RPi stuff for a while but perhaps looking in that might give someone else ideas as to ways to work around signal handling issues....

petercwallis, 2019-04-09:
Hi all,  I rebuilt a pi with a new download of pigpio in early March and am still having trouble with the JNI interface giving signal 11.  The behaviour is consistent(ish) whenever I have a separate thread running the camera (raspivid via java.lang.ProcessBuilder).  I tried swapping to
pigpioj (mattjlewis) rather than my wrapper and still no joy. Did note he has this link discussing JNI in the makefile:
http://docs.oracle.com/javase/8/docs/technotes/guides/troubleshoot/signals003.html#CIHFCAIG

This may make sense to some of you - not me.

P
ps is there a commandline way to get the version of pigpio?  - just to confirm I have downloaded the right version.

guymcswain, 2019-04-09:
> ps is there a commandline way to get the version of pigpio? - just to confirm I have downloaded the right version.

''' pigpiod -v

guymcswain, 2020-01-03:
Fixed in V65.

closed 2020-01-03 by guymcswain