joan2937/pigpio #141: gpioCfgPermissions not doing what I expect

matherp, 2017-06-30:
I'm writing in C and linking the pigpio library (not the daemon). I want to leave the i2s pins untouched so they can be used by linux so I'm calling gpioCfgPermissions before the call to gpioInitialise.

As as test I'm setting gpioCfgPermissions to zero so I shouldn't be able to touch any of the pins.

However, gpioSetMode and gpioWrite seem to work as normal and I can toggle any of the gpio

I'm obviously missing something as usual. Help appreciated

joan2937, 2017-06-30:
The documentation is at the least misleading or just plain wrong.  These permissions only affect access to the library via the socket or pipe interface.  They don't have any affect on programs linked to the C library (which in effect are within the daemon).

I will update the documentation.

matherp, 2017-06-30:
Does this mean I can never use gpio from Linux outside of my C program or just that I need to be careful not to change the settings of gpio used elsewhere?

SlySven, 2017-07-04:
I was going to say you could include the `pigpio.h` and link to the library and make your program (run as `root`) be the daemon __instead__  then it can control what other applications can access via the `pigpio_if/if2` libraries - though you should use the same lock/pid file so the original daemon and yours will not both run at the same time.  I have a self-written UPS daemon that wants exclusive access to the serial port and a single other GPIO that is reserved for the UPS to tell the RPi that the battery has nearly run out and must shutdown with 45 seconds before it switches the power off - I realise now that there is not the protection that I thought for.

I don't know but it would seem to be potentially useful if such a daemon could have a similar construct to `gpioCfgPermissions` say `gpioCfgReservations` with which it could prevent write access (via socket or other means) from other programs but which would allow unhindered access by itself to those same resources...

joan2937, 2018-02-08:
The documentation has been clarified in V65.

closed 2018-02-08 by joan2937