joan2937/pigpio #142: Using PCA9655e with PIGPIO socket interface

Joey-1970, 2017-07-06:
Hello all,

I use a PCA9655e with PIGPIO socket interface.
- Set the configuration on Register 6/7 to 16xOutput: no problem (I2CWD/57)
- Set port-value to a provided status on register 2/3: no problem (I2CWD/57)
I try to read Output on different ways (I2CRW/63 - I2CRI/67) but it failed - and this is the specific I don't understand: Not for all, the most work, but if the status of the two register (2/3) 10-0. in came back on the socket interface as 0-10. I trry different port-value: set 9-0 came back as 9-0, set 11-0 came back as 11-0 (and so on) but if I set 10-0 it came back as 0-10...
I can't unterstand what happend there. Why the result of 10-0 is turned to 0-10 over the socket interface. If I use pigs in works!

Joachim

Joey-1970, 2017-07-07:
...when I use the pigs in the command line, it seems to be okay, but not when using socket interface...
Anyone a idea??

Joachim

Joey-1970, 2017-07-08:
...a few further informations...
Here are my test-script:
`$Host = "192.168.178.25";
$Port = 8888;
$Data = pack("L*", 63, 4, 2, 0);
				
$Socket = stream_socket_client("tcp://".$Host.":".$Port, $errno, $errstr, 5);
				
stream_set_timeout($Socket, 5);
stream_socket_sendto($Socket, $Data);
$buf = fgets($Socket, 16 + 1);

$MessageArray = unpack("L*", $buf);
print_r($MessageArray);`

When the state of the PCA9655e output is something differt from 10 it is like this:
`Array
(
    [1] => 63
    [2] => 4
    [3] => 2
    [4] => 10249
)`
but if the state is 10:
`Array
(
    [1] => 63
    [2] => 4
    [3] => 2
)`
there are only three values but no error-message...
Why this is like it is???

Joachim

Joey-1970, 2017-08-02:
...I changes the PHP socket-Interface and it works.

But now there is another problem:
Set and read for a PCA9655E works. I configure it as input and use the interrupt on GPIO 17 and set the notification on it. But without nothing I get very much notifications, so I decide to set a glitch filter but I must set it to > 100ms to get only a notification when there is a change and the interrupt turn to 0 until I read the state of input.
I don't understand why are there so much notifications...is it because I use a switch mode power supply?
Wound it better to use a Noise Filter?

Joachim

Joey-1970, 2017-08-04:
...is it possible to configure the speed of sampling rate from a GPIO configure as input?

joan2937, 2017-08-07:
The GPIO sampling rate is fixed to that set when the pigpio daemon (or library) is started.  You would have to restart the daemon to set a new rate.

I can't help with the PCA9655E (is it similar to the PCA9685?).  Unless I've got or have used the hardware all I could do would be to read the datasheet, which you can do as well as me.

Joey-1970, 2017-08-19:
Hello Joan,
it was a failure in the hardware, now it works. Sometimes somethings like this happens and you search for the failure at the wrong place... ;-)

Joachim

closed 2017-08-19 by Joey-1970