joan2937/pigpio #144: Problem importing pigpio

starscr3am77, 2017-07-19:
Hi, this may not be an issue related to pigpio but I thought I'd give it a shot. I am trying to run a python program called rpotter.py and it obviously uses pigpio. I followed the instructions to install it and pigpio appears to start when I run sudo pigpiod but I get an error stating ImportError: No module named pigpio. Does my rpotter script need to be in a particular place? If anyone can assist I would greatly appreciate it. Thanks!

MrYsLab, 2017-07-19:
If you provide the code it would be easier to determine what is wrong.
From the description above it is unclear if you are starting your script and then pigpiod or vice versa.
Please clarify.

starscr3am77, 2017-07-19:
Here is the code, it crashes right at the import pigpio. My steps to run it are:

$ source ~/.profile
$ workon cv
$ sudo pigpiod
$ sudo modprobe bcm2835-v4l2
$ python2.7 rpotter.py

`#!/usr/bin/python
# -*- coding: utf-8 -*-
'''
  _\
  \
O O-O
 O O
  O
  
Raspberry Potter
Ollivander - Version 0.2 
Use your own wand or your interactive Harry Potter wands to control the IoT.  
Copyright (c) 2016 Sean O'Brien.  Permission is hereby granted, free of charge, to any person obtaining a copy of this software and associated documentation files (the "Software"), to deal in the Software without restriction, including without limitation the rights to use, copy, modify, merge, publish, distribute, sublicense, and/or sell copies of the Software, and to permit persons to whom the Software is furnished to do so, subject to the following conditions:
The above copyright notice and this permission notice shall be included in all copies or substantial portions of the Software.
THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY, FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM, OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE SOFTWARE.
'''
import io
import numpy as np
import argparse
import cv2
from cv2 import *
import picamera
import threading
import sys
import math
import time
**import pigpio**
import warnings`

MrYsLab, 2017-07-19:
If you open a python shell (in a command terminal type: python) and then type:
`
import pipgio
`
does an error comeback? If yes, check /usr/lib/python2.7/dist-packages to make sure pigpio was installed properly.

starscr3am77, 2017-07-19:
Looks like we're making progress! I got the same error, pigpio not found. I checked the dist-packages and there is no folder in there. However I did use these steps to install it. How can I redo it to ensure it gets installed in the correct directory?

`rm pigpio.zip

sudo rm -rf PIGPIO

wget abyz.co.uk/rpi/pigpio/pigpio.zip

unzip pigpio.zip

cd PIGPIO

make

sudo make install`

MrYsLab, 2017-07-19:
I don't recall if enabled this feature or not, but in rasp-config, if you select interfacing options and then REMOTE GPIO and then reboot, does that fix the problem?

If it is still not working you might want to do an apt-get update followed by and apt-get dist-upgrade.

starscr3am77, 2017-07-19:
I enabled REMOTE GPIO , ran the update and dist-upgrade. Rebooted and still get the same error. What file or folder should I be looking for in the /usr/lib/python2.7/dist-packages folder? I do see pigpio.py and pigpio.pyc but no folders. 

I also noticed there are two python2.7 folders. One in /usr/lib/python2.7 and one in /usr/local/lib/python2.7 What's the difference? I apologize for my ignorance. New to this. Also, I'm trying to run this in a virtual environment. I used this [link](http://www.pyimagesearch.com/2016/04/18/install-guide-raspberry-pi-3-raspbian-jessie-opencv-3/) to do it. Could that have something to do with why I cannot import it?

starscr3am77, 2017-07-19:
After I typed that I tested it again. I was able to import pigpio if I'm not in the virtual environment. If I am, it fails. Again, sorry to be a pain but how can I ensure the module is installed in the virtual environment?

MrYsLab, 2017-07-19:
To answer your last question first, you need to create your virtual environment with the proper flags:
virtualenv --system-site-packages my_env
The --system-site-packages is the operative part here.

Now on to your second question. The /usr/lib path is where the package is initially installed with the Raspbian distribution. When you upgrade and install pigpio, it installs into the /usr/local/lib directory.

To see where, and in what order Python searches for imports, if you open a python shell and type:
```
import sys
print(sys.path)
```
You will see the paths and their search order displayed.

My personal preference is to avoid virtual environments precisely for the problems you encountered. You can try continuing on with the virtual environment and hopefully this is the only problem you will see caused by the virtual environment.

Good luck.

starscr3am77, 2017-07-19:
Thank you so much! I looked in the virtual environments site-packages folder and saw that the pigpio files were not in there. I copied them from the /local/lib/python2.7/site-packages folder and it ran perfectly. You're a life saver!

MrYsLab, 2017-07-19:
Glad to help. You might want to close this issue ;-) if all is working now.

closed 2017-07-19 by starscr3am77