joan2937/pigpio #146: Pulse limit in sending waves

ffleandro, 2017-08-06:
I'm trying to use the [pyslinger](https://github.com/bschwind/ir-slinger/blob/master/pyslinger.py) library and experiencing some problems regarding pigpio (I think).

The original code for the library, uses pigpio C library wrapped in the Python Code, and everything works fine.
However I wanted to use the pigpio daemon instead of direct access to the C library.
I've successfully ported the code for using the native pigpio python client, however I can't seem to send waves larger than 12000 pulses.

Getting the following error log:
```
(...)
  File "/home/fhome/fresh-home-server/irslinger/pyslinger.py", line 240, in send_code
    pulses = self.gpio.wave_add_generic(self.protocol.wave_generator.pulses)
  File "/usr/local/pyenv/versions/fhserver/lib/python3.5/site-packages/pigpio.py", line 2115, in wave_add_generic
    self.sl, _PI_CMD_WVAG, 0, 0, len(pulses)*12, extents))
  File "/usr/local/pyenv/versions/fhserver/lib/python3.5/site-packages/pigpio.py", line 1001, in _pigpio_command_ext
    dummy, res = struct.unpack('12sI', sl.s.recv(16))
struct.error: unpack requires a bytes object of length 16
```

I've sent smaller waves with no problem, but always get that error when sending larger than 12000 pulses.
I suspect some variable is getting overflowed or I'm passing some limit in the communication with the pigpiod daemon service.

I've successfully sent the same wave using the original library (direct access to libpigpio.so in the Python code), but with no success using the client/socket communication between pigpio python client and pigpiod daemon.

Could someone share some insights before I start to dig deeper into pigpio's code?

ffleandro, 2017-08-06:
I guess I can break the wave into two separate `wave_add_generic` calls, but it would be interesting to know what limit are we breaking into.

joan2937, 2017-08-07:
You have reached a hard limit.

In `pigpio.h`

```C
#define PI_WAVE_BLOCKS     4
#define PI_WAVE_MAX_PULSES (PI_WAVE_BLOCKS * 3000)
#define PI_WAVE_MAX_CHARS  (PI_WAVE_BLOCKS *  300)
```

So the maximum number of pulses (for all defined waves added together) is 12000.

I'd assume this is a hard hard limit.  Changing `PI_WAVE_BLOCKS` to another value will probably damage other functionality.  You'd need to check very carefully if you take that route.

Another hard limit you might reach using the daemon is the maximum socket message size of (I think) 65k bytes.  Again I would treat that as a hard hard limit.

ffleandro, 2017-08-07:
Ok, thank you for the explanation.

>So the maximum number of pulses (for all defined waves added together) is 12000

So does that mean that 2 separate calls to `add_wave_generic` would generate the same error?

Then, why does the original code works with no error (also no communication between client/socket)?

I tried separating into two calls, but the wave is not being appended, it's being merged.

From the documentation:
>If the added waveform is intended to start after or within the existing waveform then the first pulse should consist solely of a delay.

What exactly does that mean?

`pigpio.pulse(0,0,100)` adding a 100uS delay for instance?

joan2937, 2017-08-07:
If you don't have a delay the new pulses will be relative to the wave start (time 0).   If the time of any new pulse corresponds to an existing pulse then the existing pulse will be reused (by updating the GPIO on/off bits).

Your example is correct, if the new pulses are to continue after the existing pulses add the appropriate delay as the first new pulse.

You can find the length of the current wave with [wave_get_micros](http://abyz.co.uk/rpi/pigpio/python.html#wave_get_micros) which may be used as the delay.

ffleandro, 2017-08-07:
Ok, understood. Will post my results.

How about the first question, will two separate calls to `add_wave_generic` hit the hard limit I was experiencing with waves larger than 12000 pulses?

joan2937, 2017-08-07:
Yes, there is only space for 12000 pulses, it doesn't matter if they are added all in one go or over a series of calls.

ffleandro, 2017-08-07:
So my only alternative is chaining two distinct waves?

joan2937, 2017-08-07:
You would need to delete the old wave before creating the new one - so I wouldn't call that chaining.

Have a look at [RPIO.GPIO](https://github.com/metachris/RPIO) which generates pulses in a different way.  It may do what you want.  I don't know if it will, I just know it works differently.

ffleandro, 2017-08-07:
So I finally got home and was intrigued by why the original `pyslinger` library didn't break the limit, so I tried separating the `wave_add_generic` into two separate calls and my scenario is working correctly.

I guess I was breaking the limit of the maximum socket message size instead of the 12000 pulses.

Currently my scenario doesn't need 12k pulses (I'm sending an AC Infrared command), but if I do need a bigger command (currently all my cases use 10k to 11k) is there any possibility to increase this 12k limit or some other workaround?

I could try another GPIO library as you mentioned, but the client/socket approach you used in `pigpio` is a must have in my current Architecture.

treflip, 2017-12-20:
how are you creating these long pulse counts? I am trying to generate a 20 min signal and was thinking of using the file read functionality. I am not entirely sure how the file needs to be formatted nor how process it afterwards....

ffleandro, 2017-12-20:
I'm using [pyslinger](https://github.com/bschwind/ir-slinger).
Basically I pass a string of 1's and 0's according to my infrared AC command codes.

Each 1 represents a pause and a long burst of the infrared LED and a 0 represent a pause and a small burst.

closed 2017-12-20 by ffleandro