joan2937/pigpio #147: Setting pull-up via pigs strange behavior

MaitreDede, 2017-08-10:
Hi,
Using a Pi3 with raspbian, I use pigs to do some gpio tests.
The script I try to use :
```
pigs m 25 0 m 24 0 m 23 0 m 18 0
pigs pud 25 u pud 24 u pud 23 u pud 18 u
pigs r 25 r 24 r 23 r 18
```
In words : set gpios 18,23,24,25 as input, with pull-up, and read them.

But when I try to execute these commands, the first "works", but it has corrupted something in my pi, since after the first, I can't execute anything more : `-bash: pigs: command not found` or any other things I may try.
The only solution : power off and reboot.

If I split each command, the corruption occurs when I execute :
```
pigs m 25 0 # ok
pigs m 24 0 # ok
pigs m 23 0 # error  -bash: /usr/bin/pigs: Input/output error
```

Any idea ?

MaitreDede, 2017-08-10:
Note : I don't have this problem when executing with socket interface.

joan2937, 2017-08-10:
The command to set a GPIO as an input is `pigs m 23 r` (see [pigs](http://abyz.co.uk/rpi/pigpio/pigs.html#M/MODES)).

`pigs m 23 0` will set the GPIO to mode ALT0.

A number of modes are marked `reserved` for certain GPIO.  ALT0 is a reserved mode for GPIO 16, 17, 22-27.

MaitreDede, 2017-08-10:
Thanks. Sometimes I may not do things when it is late...

closed 2017-08-10 by MaitreDede