joan2937/pigpio #149: Using pigpio for MCP4725 and TSL2951 on I2C

rolly-ng, 2017-08-19:
Dear Joan,

Nice to meet you and this is Rolly from CityU of Hong Kong.

I am working on a research project that using light sensor to detect trace lead ions in drinking water and I am using Raspberry Pi3 as MCU. My setup has a Adafruit TSL2951 light sensor together with a Adafruit MCP4725 DAC.

I would like to incorporate the pigpio Python library for my setup and I would like to ask for your kind advices on the pigpio:

1) I am trying to get a stable 2 kHz square wave on the MCP4275 since I need to vary the output voltage between 0 to 5Vpp. Can I do that with the pigpio Python? At the moment with the Adafruit library, it swings from 1.7 kHz to 2.1 kHz which is not stable.

https://www.raspberrypi.org/forums/viewtopic.php?t=190134

2) If I pair both DAC and the light sensor to the default I2C channel (pin03 and 05), would it be slowing down the MCP4275 and blocking the TSL2951 from capturing light? I like to have simultaneous square wave generation and light recording at the same instance. Is there something called bit banging to use spare GPIO pins for I2C communication?

3) The maximum I2C baud rate supported by the two devices are different. MCP4275 to 3.4 Mbps, TSL2591 to 400 Kbps only. With 1.7 Mbps on I2C, I can get the unstable 2 kHz, but with 400 Kbps, this is impossible with the Adafruit Python library. Should I must use bit banging to split the I2C operation via pigpio?

I am attaching the specification of both TSL2951 and MCP4275 for your quick reference.

Thank you very much for your kind attention!

Best regards,

Rolly

electrokean, 2017-08-20:
@rolly-ng I've used TSL2651 (older chip) with pigpio and had no problems (4 at once on separate bit-banged I2C buses), but I do suspect you will have trouble generating a square wave with constant frequency via DAC I2C commands alone, especially when interleaved with reading the light sensor.
Just a thought - I presume you are using a DAC because you need to occasionally vary the max amplitude of the square wave.  Could you also use an analog switch or P-Ch FET to switch the output of the DAC on/off, with pigpio generating the 2kHz gating signal on a separate GPIO.  If you need to switch between two levels, then maybe two DACs plus a SPDT analog switch.
You could also put the TSL2951 on a bit-banged I2C bus run at a slower speed to separate the devices - in fact that is what I would try first, but I think the above it more elegant if it suits your measurement requirements.

rolly-ng, 2017-08-20:
@electrokean Thank you for the kind suggestion. I need to vary the 2kHz for 4 different voltages, e.g. 0, 1.5, 2.5, 4.5 Vpp. But adding 4 DACs to my setup is somehow complicated. Besides the MCP4725, do you think the MCP4921 via the SPI interface should address my problem? So I can have the I2C dedicated to the TSL2951 and the SPI to the MCP4921, can this be done on the Raspberry Pi3 via pigpio Python?
Thanks!

electrokean, 2017-08-20:
Yes, I think an SPI DAC will work a lot better than the I2C one in that case.

rolly-ng, 2017-08-28:
@electrokean @joan2937 

I have hooked up a MCP4921 to my RPi3 and these are my connections:
VDD to 5V
CS to GPIO8
SCK to GPIO11
SDI to GPIO10
LDAC to ground
VREF to 5V
AVSS to ground
VOUT to oscilloscope

Then I modified the following python code and I get a much stabilized 2 kHz pulse train.
`import spidev
from time import sleep
#import RPi.GPIO as GPIO
DEBUG = False

# setup and open an SPI channel
spi_max_speed = 32 * 1000000 # 4 MHz
spi = spidev.SpiDev()
spi.open(0,0)
spi.max_speed_hz = spi_max_speed

def setOutput(val):
    # lowbyte has 8 data bits D0..D7
	lowByte = val & 0xff; # 0b 1111 1111
    # highbyte has control and data bits
    # control bits are:
    # B7, B6,   B5, B4,     B3,  B2,  B1,  B0
    # W  ,BUF, !GA, !SHDN,  B11,  B10,  B9,  B8
    # B7=0:write to DAC, B6=0:unbuffered, B5=1:Gain=1X, B4=1:Output is active
	highByte = ((val >> 8) & 0xff) | 0b0 << 7 | 0b0 << 6 | 0b1 << 5 | 0b1 << 4;
    # by using spi.xfer2(), the CS is released after each block, transferring the
    # value to the output pin.
	spi.xfer2([highByte, lowByte])

try:
    print "Expect 2kHz square wave by MCP4921..."
    print "Ctrl+C to quit"

    while(True):
            newbits = int(0) #min. Vout
            setOutput(newbits)
            sleep(0.000125)
            newbits = int(2376) #max. Vout
            setOutput(newbits)
            sleep(0.000125)

except KeyboardInterrupt:
    print "\nClosing SPI channel"
    spi.close()

def main():
    pass

if __name__ == '__main__':
    main()`

Now, I would like to use pigpio to access DMA for even better stabilization. I found several bit-bang 
SPI examples for 12 bit ADC here but none for DAC, http://abyz.co.uk/rpi/pigpio/examples.html#C code

But how can I modify it to work on my 12 bit DAC?

I observed that the MCP4921 with the following write commands, but I am not able to modify the examples. Could you please help?

`   /*
   MCP4921 12-bit DAC

   1  2  3  4  5   6   7  8  9  10 11 12 13 14 15 16
   AB BF GA SD B11 B10 B9 B8 B7 B6 B5 B4 B3 B2 B1 B0

   AB  0  0=write to DAC
   BF  0  0=unbuffered 1=buffered
   GA  1  1=gain 1x
   SD  1  1=output is active
   */`

Thank you so much,
Rolly

closed 2017-09-07 by rolly-ng