joan2937/pigpio #150: waveAddSerial does not align data of integer multiple baud rates

guymcswain, 2017-08-23:
Overlapping data of same baud rate is aligned:
`$ pigs wvnew wvsc 0
0
$ pigs wvas 25 19200 8 2 0 0x55 0x55 0x55 0x55
41
$pigs wvsc 0
82
$ pigs wvas 17 19200 8 2 0 0x55 0x55 0x55 0x55
41
$pigs wvsc 0
82
`

Overlapping data of integer multiple baud rates do not align (higher consumption of DMA control blocks):
`$pigs wvnew wvsc 0
0
$pigs wvas 25 19200 8 2 0 0x55 0x55 0x55 0x55
41
$pigs wvsc 0 wvsm 0
82
2136
$pigs wvas 17 9600 8 2 0 0x55 0x55
47
$pigs wvsc 0 wvsm 0
102
2136`

In theory, this second test should have had the same results as the first with all 20 bit transitions at 9600 baud aligning to the 40 bit transitions at 19200.   I took a quick look at the C source and wonder if the bit delay rounding to nearest 1/2 microsecond in function 'waveBitDelay' could explain this behavior.

My desire (requirement) is to generate multiple serial data streams as efficiently as possible using 'standard' baud rates.

joan2937, 2017-08-25:
I would need to carry out some tests to confirm your findings, but that's not something I will be doing in the near future.

At the moment you will have to accept the current behaviour although I would prefer the behaviour you expect.

guymcswain, 2017-09-15:
This issue seems a bit esoteric so I'm closing with comments on my findings.

A multi-channel arbitrary waveform will requires up to 3 CBs per pulse:  One CB per delay, one for gpios rising and one for gpios falling.  Rounding of the delay is not the issue.   The following illustrates this point:
```
$pigs wvnew wvsc 0
0
$pigs wvas 17 19200 8 2 400 0x55 0x55 0x55 0x55
40 (pulse count)
$pigs wvas 25 19200 8 2 400 0xaa 0xaa 0xaa 0xaa
40 (pulse count)
$pigs wvsc 0 wvsm 0
107 (CBs)
2428 (micros)
```
Tally of CBs:
4 start bits and 4 stop bits have gpios with same polarity so 8 pulses x 2 CBs = 16 CBs
7 bits of each byte have gpios with opposing polarity or 4 x 7 x 3 = 84 CBs
The lsb of each byte is in phase with start bit so 4 x 2 = 8 CBs
Total CBs = 108 (vs 107, oh well, close enough)

PS:  The results shown above were taken from a hacked version of pigpio.c where the leading start pulse and trailing stop pulse were removed from the` wvas` (gpioAddSerial) command.  You must provide an offset greater than the largest bit period (400 used here) to get the waveforms to align.

closed 2017-09-15 by guymcswain