joan2937/pigpio #151: RPi3 internal wifi offline after pigpiod start usage from python

bdn76, 2017-08-25:
I'm use pigpiod for spi/i2c communication from python but RPi3 disconnectoned from wifi immediatly after my script start. I'm remove pigpiod related code and wifi on RPi3 work fine. Copper ethernet work fine in both cases.

joan2937, 2017-08-25:
I only use wifi and have never seen this behaviour.

Details of the set-up are needed.

bdn76, 2017-08-25:
I'm sorry. /etc/network/interfaces issue.

bdn76, 2017-08-25:
I'm can reproduce issue. Bug code:

#!/usr/bin/python3

import pigpio
from time import sleep

def bug():
    en = {'y': {'cs': 18, 'miso': 23, 'mosi': 24, 'clk': 25, 'baud': 100000, 'mode': 1},
         'x': {'cs': 12, 'miso': 16, 'mosi': 20, 'clk': 21, 'baud': 100000, 'mode': 1}}
    mo = {'y': {'step': 4, 'dir': 3, 'enable': 2, 'alarm': 17, 'freq': 65536},
         'x': {'step': 6, 'dir': 13, 'enable': 19, 'alarm': 26, 'freq': 65536}}
    pigpio.exceptions = False
    pi = pigpio.pi()
    for e in en.keys():
        pi.bb_spi_open(en[e]['cs'], en[e]['miso'], en[e]['mosi'], en[e]['clk'], en[e]['baud'], en[e]['mode'])
    for m in mo.keys():
        pi.write(mo[m]['enable'], 0)
        pi.hardware_clock(mo[m]['step'], 0)
        pi.write(mo[m]['dir'], 0)
        pi.set_pull_up_down(mo[m]['alarm'], pigpio.PUD_DOWN)

if __name__ == "__main__":
    bug()

Env on Rpi3:

root@pi3:~# cat /etc/debian_version 
9.1
root@pi3:~# uname -a
Linux pi3 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l GNU/Linux
root@pi3:~# dpkg -l | grep pigpio
ii  pigpio                          1.60-1                       armhf        Library for Raspberry Pi GPIO control
ii  python-pigpio                   1.60-1                       all          Python module which talks to the pigpio daemon (Python 2)
ii  python3-pigpio                  1.60-1                       all          Python module which talks to the pigpio daemon (Python 3)
root@pi3:~#

joan2937, 2017-08-25:
The problem is caused when clock 2 (attached to GPIO 6) is switched off (set to 0).

This doesn't seem to be a bug within pigpio.

I have no idea if this is new behaviour.  Perhaps the system now uses clock 2 as well as or instead of clock 1.

All I can suggest is you don't switch the clock 2 off.

bdn76, 2017-08-26:
```
Yes, you are right. This code:

#!/usr/bin/python3
import pigpio

pi = pigpio.pi()
pi.hardware_clock(6, 0)

or this command:

pi@pi3:~ $ pigs hc 6 0

make a brick from RPi3 connected by wifi.

Why set hc to 0 kill wifi but default value is not? Am i see any signal on pin 6 immediatly after restart? 
```

joan2937, 2018-02-13:
I am closing this as there is no evidence of a fault within pigpio.   Setting a GPIO to an incorrect mode can crash the Pi.

If you think a particular usage may be common and will cause a crash let me know and I will consider adding a warning.

closed 2018-02-13 by joan2937