joan2937/pigpio #152: PIGPIO-Daemon and Hifiberry AMP+

Joey-1970, 2017-09-02:
...maybe a important information:
If you want to use a Hifiberry AMP+ (or maybe other models), it stop working if you start the pigpio daemon (with sudo pigpiod).

I don't know why, but it seemd to be impossible to work with both features on one Raspberry Pi...

Joachim

guymcswain, 2017-09-02:
From the FAQ:
> By default pigpio uses the PCM peripheral leaving the PWM peripheral free for medium quality audio.
You can change the default with a configuration option.  For C use gpioCfgClock, for the pigpio daemon use the -t option.

Have you tried this?

Joey-1970, 2017-09-04:
Hello guymcswain,

I tried it and it don't work because my modules use Hardware PWM for RGB(W).
> Default PCM. pigpio uses one or both of PCM and PWM. If PCM is used then PWM is available for audio. If PWM is used then PCM is available for audio. If waves or hardware PWM are used neither PWM nor PCM will be available for audio.
> 
But the reason for posting it, was more that the "normal" user will use the feature from the raspi-config to enable the GPIO-Server and this starts without any options.
To use PIGPIO together with some audio HAT needed minimum to things:
- no use of hardware PWM
- start with option -t0 (1 is default)
I suspect that the most "hobby-user" will don't know it...

Joachim

guymcswain, 2017-09-04:
This makes sense.  I haven't used audio.  Thanks for the info.

guymcswain, 2020-01-02:
This is a known limitation of the pigpio library:  PCM or PWM peripheral is needed to run the DMA timing.  If audio is used, you must use PWM for DMA timing.  If your application needs both audio and PWM then you cannot use pigpio (except possibly with severe restrictions such as turning off alerts along with never using waveforms or bit-bang APIs, and probably more).

closed 2020-01-02 by guymcswain