joan2937/pigpio #153: Concurrent DAC and ADC using pigpio?

rolly-ng, 2017-09-07:
Dear all,
I was asking questions on how to generate stable 2 kHz square pulses on MCP4725 DAC via I2C.
[https://github.com/joan2937/pigpio/issues/149](url)

I must thank the forum members for their useful suggestions and my solution was to change to MCP4921 DAC via SPI. I have also changed from Python to C using Joan's pigpio library, so I can produce stable 2 kHz square pulses as I boot into CLI. If I boot into GUI, it also works but the frequency drifts if I drag/open/close any window. But I am satisfy on CLI, so it is okay.

However, I also need to capture light signals continuously from a Adafruit TSL2591 ADC via I2C concurrently. I am surprised that pigpio only allows a single instance running. So, if I have the DAC to produce the square pulses, I cannot run the ADC at the same time. I have to kill the DAC in order to run the ADC, but this is not acceptable.

Can anyone advise how to run the DAC on SPI and ADC on I2C concurrently using the pigpio?

I am including my two codes below, 1) MCP4921 DAC-SPI, 2) TSL2591 ADC-I2C.

Thank you very much!
Rolly

`#include <stdio.h>
#include <sys/ioctl.h>
#include <stdlib.h>
#include <pigpio.h>

#define B_SIZE 128

void DAC_Write(int h, int value)
{
   char buf[2];
   char chanBits, dataBits;

   chanBits = (((value >> 8) & 0xFF) | 0b0 << 7 | 0b0 << 6 | 0b1 << 5 | 0b1 << 4);
   dataBits = ( value & 0xFF);

   buf[0] = chanBits;
   buf[1] = dataBits;

   spiXfer(h, buf, NULL, 2);
}

int main(int argc, char *argv[])
{
   int i;
   int buffer[B_SIZE];
   int dac;

   if (gpioInitialise() < 0) return 1;

   dac = spiOpen(0, 4000000, 0);

   for (i = 0; i < B_SIZE; ++i)
   {
      if (i < 63) {
      buffer[i] = 0;
	  }
      else {
      buffer[i] = 4095;
	  }
   }

   for (;;)
   {
      for (i = 0; i < B_SIZE; ++i)
      {
         DAC_Write(dac, buffer[i]);
      }
   }

   return 0;
}`

`#include <stdio.h>
#include <stdlib.h>
#include <unistd.h> 
#include <sys/ioctl.h>
#include <pigpio.h>

// ALL COMMAND TSL2591
#define TSL2591_VISIBLE           (2)       // channel 0 - channel 1
#define TSL2591_INFRARED          (1)       // channel 1
#define TSL2591_FULLSPECTRUM      (0)       // channel 0

#define TSL2591_ADDR              (0x29)
#define TSL2591_READBIT           (0x01)

#define TSL2591_COMMAND_BIT       (0xA0)    // bits 7 and 5 for 'command normal'
#define TSL2591_CLEAR_BIT         (0x40)    // Clears any pending interrupt (write 1 to clear)
#define TSL2591_WORD_BIT          (0x20)    // 1 = read/write word (rather than byte)
#define TSL2591_BLOCK_BIT         (0x10)    // 1 = using block read/write

#define TSL2591_ENABLE_POWERON    (0x01)
#define TSL2591_ENABLE_POWEROFF   (0x00)
#define TSL2591_ENABLE_AEN        (0x02)
#define TSL2591_ENABLE_AIEN       (0x10)

#define TSL2591_CONTROL_RESET     (0x80)

#define TSL2591_LUX_DF            (408.0F)
#define TSL2591_LUX_COEFB         (1.64F)   // CH0 coefficient
#define TSL2591_LUX_COEFC         (0.59F)   // CH1 coefficient A
#define TSL2591_LUX_COEFD         (0.86F)   // CH2 coefficient B

enum
{
  TSL2591_REGISTER_ENABLE           = 0x00,
  TSL2591_REGISTER_CONTROL          = 0x01,
  TSL2591_REGISTER_THRESHHOLDL_LOW  = 0x02,
  TSL2591_REGISTER_THRESHHOLDL_HIGH = 0x03,
  TSL2591_REGISTER_THRESHHOLDH_LOW  = 0x04,
  TSL2591_REGISTER_THRESHHOLDH_HIGH = 0x05,
  TSL2591_REGISTER_INTERRUPT        = 0x06,
  TSL2591_REGISTER_CRC              = 0x08,
  TSL2591_REGISTER_ID               = 0x0A,
  TSL2591_REGISTER_CHAN0_LOW        = 0x14,
  TSL2591_REGISTER_CHAN0_HIGH       = 0x15,
  TSL2591_REGISTER_CHAN1_LOW        = 0x16,
  TSL2591_REGISTER_CHAN1_HIGH       = 0x17
};

typedef enum
{
  TSL2591_INTEGRATIONTIME_100MS     = 0x00,
  TSL2591_INTEGRATIONTIME_200MS     = 0x01,
  TSL2591_INTEGRATIONTIME_300MS     = 0x02,
  TSL2591_INTEGRATIONTIME_400MS     = 0x03,
  TSL2591_INTEGRATIONTIME_500MS     = 0x04,
  TSL2591_INTEGRATIONTIME_600MS     = 0x05,
}tsl2591IntegrationTime_t;

typedef enum
{
  TSL2591_GAIN_LOW                  = 0x00,    // low gain (1x)
  TSL2591_GAIN_MED                  = 0x10,    // medium gain (25x)
  TSL2591_GAIN_HIGH                 = 0x20,    // medium gain (428x)
  TSL2591_GAIN_MAX                  = 0x30,    // max gain (9876x)
}tsl2591Gain_t;

//Delay getFullLuminosity function
#define LUMDELAY 100000

int getFullLuminosity(void){
	int fd = 0;
	uint32_t vis_ir_0, vis_ir_1;
	i2cWriteByteData(fd, TSL2591_COMMAND_BIT | TSL2591_REGISTER_ENABLE, TSL2591_ENABLE_POWERON | TSL2591_ENABLE_AEN | TSL2591_ENABLE_AIEN); //enable the device
	i2cWriteByteData(fd, TSL2591_COMMAND_BIT | TSL2591_REGISTER_CONTROL, TSL2591_INTEGRATIONTIME_100MS | TSL2591_GAIN_LOW); //set gain and timing
	//Wait for the conversion to complete
	gpioDelay(LUMDELAY);
	//Reads visible + IR diode from the I2C device auto
	vis_ir_1 = i2cReadWordData(fd, TSL2591_COMMAND_BIT | TSL2591_REGISTER_CHAN1_LOW);
	vis_ir_1 <<= 16;
	vis_ir_0 = i2cReadWordData(fd, TSL2591_COMMAND_BIT | TSL2591_REGISTER_CHAN0_LOW);
	vis_ir_1 |= vis_ir_0;
	i2cWriteWordData(fd, TSL2591_COMMAND_BIT | TSL2591_REGISTER_ENABLE, TSL2591_ENABLE_POWEROFF); //disable the device
	return vis_ir_1;
}

int main(int argc, char *argv[])
{
    uint16_t ch0, ch1;
    uint32_t visible_ir;
    
    if (gpioInitialise() < 0) return 1;
    
    int fd = i2cOpen(1, TSL2591_ADDR, 0);
       
	while(1)
	{
		visible_ir = getFullLuminosity();
	    // Reads two byte value from channel 0 (visible + infrared)
	    ch0 = (visible_ir & 0xFFFF);
	    // Reads two byte value from channel 1 (infrared)
	    ch1 = (visible_ir >> 16);
	    //printf("TSL2591 Full Spectrum: %zu\n",ch0);
	    //printf("TSL2591 Infrared: %zu\n",ch1);
	    if ( ch0 >= ch1 )
	    	printf("TSL2591 Visible: %zu\n",ch0-ch1);
	    sleep(1);
	}
	i2cClose(fd);
	return 0;
}`

guymcswain, 2017-09-07:
Maybe a silly question but why did you abandon Python?  If you use the pigpio Python library you should be able to run both devices from either a single Python script or, I believe it should be possible, to have two independent scripts talking to pigpiod - one running your DAC, the other running TSL*.  This is a client-server architecture and why I like to use the socket interface.

This is for the case where you are accessing the Broadcom peripherals via pigpio.  If you wish to use bit-bang 'devices' then you'll run into the problem of the waveform generator as a shared resource :)

rolly-ng, 2017-09-08:
@guymcswain 
Hi, I changed to C because while I look at the DAC output with an oscilloscope, the square pulse frequency is unstable compared to those produced by C. With Python, it drift from 1.5 to 2.5 kHz, but with C it is much better from 2.05 to 2.15 kHz.

Anyway, I found my solution. I have to put the spiXfer loop in the DAC_Write_Pulse function, not inside the main() in order to call the ADC. It works now and case closed.

Thanks to Joan for the kind advises.

Rolly

closed 2017-09-08 by rolly-ng