joan2937/pigpio #154: BSPIO without MISO pin

MaitreDede, 2017-09-15:
Hi,

I would like to know if it is possible to use BSPIO command to bit-bang on GPIO without having a MISO pin.
I am driving a [TLC5947](https://www.adafruit.com/product/1429) through socket interface. The Adafruit driver is direct bit-bang, but really slow when using a code with same behavior (manual gpio writing). So I am using the BSPI* commands to achieve the same, but really faster.

If it is not in the current code, do you think it can be added ? The data read from BSPIX could return 0. This may avoid wasting gpios just for this.

Thanks.

guymcswain, 2017-09-18:
Is the problem that you have run out of pins but need to add another SPI port?  You can multiplex MOSI, MISO and SCLK by using a unique CS pin for each port.  If nothing is attached to MISO you could tie it off externally or maybe set the pullup/down after you have opened the port.

If instead you are wanting to transmit on all SPI ports in parallel, it looks like a possible work-around is to specify the same MISO pin for each SPI port.  A quick look at the code suggests MISO must only be exclusive to the other SPI pins (MOSI, SCLK and CS) of the same port.  It may be worth giving this a try.

guymcswain, 2020-01-02:
@MaitreDede , If you are still interested in this I'll review your proposal for implementing it.  You are an experienced dev and I'd like to see you contribute to this library.

guymcswain, 2020-01-25:
Upon review of its specification, the TLC5947 is not a SPI interface.  Therefore what is requested is a custom API and I don't see that as being in the broader interest of the pigpio community.

closed 2020-01-25 by guymcswain