joan2937/pigpio #156: pigpio.initialize() doenst do its job after pigpio.terminate()

joshi11, 2017-09-17:
I have a situation in which I use pigpio in Node and in native C 
Thus I need to call pigpio.terminate() in Node before I can call pigpio in C otherwise C complains about not getting the lock on the pid.
When I then try to call pigpio.initialize in Node again it doesnt seem to work since I get the message:
> gpioSetMode: pigpio uninitialised, call [gpioInitialise()]

`
var initialized;
var pigpio = require('pigpio')
var Gpio=pigpio.Gpio

        exports.test = function(led_nr, dutyCycle) {	//led_nr 0-3
		if (!initialized)	{
			console.log("calling initialize");
			pigpio.initialize(); // pigpio C library initialized here
			initialized=1;
			console.log('Hardware Revision: ' + pigpio.hardwareRevision().toString(16));
			test_led(led_nr, dutyCycle);   
		}
		else 
			test_led(led_nr, dutyCycle);    //called function to make sure I dont get race conditions
		
		
	};	

	exports.terminate = function() {
		console.log("initialized= " + initialized);
		if (initialized)	{
			pigpio.terminate(); // pigpio C library terminated here
			initialized=0;
			console.log("initialized= " + initialized);
			console.log('Terminated');
		}	//end if
	};//end terminate
`

guymcswain, 2017-09-17:
What is the return value of `pigpio.initialize()`?

You should expect to get the pigpio revision returned.  (ie V64)
`pigpio.hardwareRevision()` will still return successfully even when pigpio is not initialized.

joshi11, 2017-09-18:
Since I dont know any better I tried this:
`var res=pigpio.initialize(); // pigpio C library initialized here`
`			console.log("initialize resulte = " + res);`

With the Result:

> 
> > calling initialize
> initialize resulte = undefined
> Hardware Revision: 9000c1
> initialized= 1

So I would assume that pigpio.initialize() doesnt have a return value 

Versions:

>   pigpio: "_id": "pigpio@0.6.1",
node  v6.11.3

joshi11, 2017-09-18:
It was an issue with the wrapper - Solved!
Sorry for opening an issue at the wrong place!

closed 2017-09-18 by joshi11