joan2937/pigpio #158: Exiting C program kills PCM clock even when using PWM clock in pigpio

matherp, 2017-09-19:
I need to be able to have i2s audio running whilst starting and stopping a pigpio C I/F program.
I'm setting pigpio to use the PWM clock with a call to gpioCfgClock and I can run the program without it causing a problem with the ongoing i2s output which has to use the PCM clock. 
However, when I terminate the pigpio C I/F program the playback stops. It looks like gpioTerminate() is affecting the PCM clock even when pigpio is set to use the PWM clock.

joan2937, 2018-03-28:
Sorry for the late response.

I have looked through the code and can't find any incorrect access to the PCM peripheral.

Do you have a small C program which demonstrates the problem?

guymcswain, 2020-01-02:
@matherp , If you can provide the steps necessary to reproduce the issue we will look into it.  It appears you are using the socket interface from a C program.  If so, how did you set up the daemon?

HorstBaerbel, 2020-03-16:
I have this problem here on a RPi Zero-W with an I2S codec attached (Adafruit speaker bonnet at GPIO 18, 19, 21). Sound plays fine through ALSA until I run a C++ program that simple initializes pigpio (just gpioInitialise() is enough to distort the sound). Any ideas?

guymcswain, 2020-03-17:
Please read the FAQ. 
 http://abyz.me.uk/rpi/pigpio/faq.html#Sound_isnt_working

HorstBaerbel, 2020-03-17:
Thanks. Hadn't stumbled over that one yet...

closed 2020-03-22 by guymcswain