joan2937/pigpio #160: I2C slave only can read out 16 bytes at one read cycle. How can I increase it?

ayeteng, 2017-09-21:
I set the callback function as below and print out the received counter:

def i2c(id, tick):
    global pi
    s, b, d = pi.bsc_i2c(I2C_ADDR)
    if b:
         print(b)

I found I only can get up to 16 bytes when transmit more than 16 bytes. How can I increase it?

ayeteng, 2017-09-26:
Describe in detail about the connection:
1. Connect SDA(pin 3) to GPIO18 (pin 12), SCL(pin 5) to GPIO19 (pin 35).
2. Add 4.7K pull up resistor for both SDA and SCL. 
3. Run bsc_arduino.py but modify the i2c function below:
`def i2c(id, tick):
global pi
s, b, d = pi.bsc_i2c(I2C_ADDR)
if b:
    print(b)`

Check the I2C response by i2cdetect
root@raspberrypi:~/sam# i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- 0a -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

send the I2C packet:
root@raspberrypi:~/sam# i2cset -y 1 0xa 0x0 0x1 0x2 0x3 0x4 0x5 0x6 0x7 0x8 0x9 0xa 0xb 0xc 0xd 0xe 0xf 0x10 0x11 i

In the receive side, I still only get 16 bytes.

Anyone who can help to extend the receive buffer to 64 or 128?

ayeteng, 2017-10-05:
I did a modification and get some progress.
modify the #define BSC_FIFO_SIZE from 16 to 256
After I run my test script, I can get better result but not stable.

**Terminal 1:**
```
root@raspberrypi:~/sam# i2cset -y 1 0x13 0x60 0x61 0x62 0x63 0x64 0x65 0x66 0x67 0x68 0x69 0x6a 0x6b 0x6c 0x6d 0x6e 0x6f 0x70 0x71 0x72 0x73 0x74 0x75 0x76 0x77 0x78 0x79 0x7a 0x7b 0x7c 0x7d 0x7e 0x7f 0x80 i
root@raspberrypi:~/sam# i2cset -y 1 0x13 0x60 0x61 0x62 0x63 0x64 0x65 0x66 0x67 0x68 0x69 0x6a 0x6b 0x6c 0x6d 0x6e 0x6f 0x70 0x71 0x72 0x73 0x74 0x75 0x76 0x77 0x78 0x79 0x7a 0x7b 0x7c 0x7d 0x7e 0x7f 0x80 i
root@raspberrypi:~/sam# i2cset -y 1 0x13 0x60 0x61 0x62 0x63 0x64 0x65 0x66 0x67 0x68 0x69 0x6a 0x6b 0x6c 0x6d 0x6e 0x6f 0x70 0x71 0x72 0x73 0x74 0x75 0x76 0x77 0x78 0x79 0x7a 0x7b 0x7c 0x7d 0x7e 0x7f 0x80 i
root@raspberrypi:~/sam# i2cset -y 1 0x13 0x60 0x61 0x62 0x63 0x64 0x65 0x66 0x67 0x68 0x69 0x6a 0x6b 0x6c 0x6d 0x6e 0x6f 0x70 0x71 0x72 0x73 0x74 0x75 0x76 0x77 0x78 0x79 0x7a 0x7b 0x7c 0x7d 0x7e 0x7f 0x80 i
root@raspberrypi:~/sam# i2cset -y 1 0x13 0x60 0x61 0x62 0x63 0x64 0x65 0x66 0x67 0x68 0x69 0x6a 0x6b 0x6c 0x6d 0x6e 0x6f 0x70 0x71 0x72 0x73 0x74 0x75 0x76 0x77 0x78 0x79 0x7a 0x7b 0x7c 0x7d 0x7e 0x7f 0x80 i
```

**Terminal 2 running with my script:**

```
Got it! Status: 0x00012, byte count: 33

Receivie:
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f
10: 70 71 72 73 74 75 76 77 78 79 7a 7b 7c 7d 7e 7f
20: 80

Got it! Status: 0x00012, byte count: 29

Receivie:
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f
10: 74 75 76 77 78 79 7a 7b 7c 7d 7e 7f 80

Got it! Status: 0x00012, byte count: 33

Receivie:
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f
10: 70 71 72 73 74 75 76 77 78 79 7a 7b 7c 7d 7e 7f
20: 80

Got it! Status: 0x00012, byte count: 29

Receivie:
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f
10: 74 75 76 77 78 79 7a 7b 7c 7d 7e 7f 80

Got it! Status: 0x00012, byte count: 33

Receivie:
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f
10: 70 71 72 73 74 75 76 77 78 79 7a 7b 7c 7d 7e 7f
20: 80
```

[bsc_test.py.txt](https://github.com/joan2937/pigpio/files/1358148/bsc_test.py.txt)

Is this the limitation of the hardware or we can tweak the library?

joan2937, 2018-02-08:
I think the FIFO size is 32 bytes.  I'm not sure why I use a limit of 16 rather than 32.

See page 160 onwards of https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2835/BCM2835-ARM-Peripherals.pdf

I probably won't be doing any tests in this area in the near future.

If you are confident that BSC_FIFO_SIZE should be 32 I am happy to make that change.

joan2937, 2018-03-11:
I have had a brief look.  The hardware RX and TX FIFOs do appear to be 16 bytes.  So you can only guarantee 16 byte messages will be sent/received.

If the system can react quickly enough to the first received data it can empty the FIFOs at a quick enough pace to allow arbitrary message lengths.  That is a big if and can not be guaranteed.

The receive side isn't much of a problem.  I'll have to think some more about the transmit side.

joan2937, 2018-03-13:
I have changed the buffer size to 512 bytes.

However the hardware RX/TX FIFO is 16 bytes so you can only guarantee messages of 16 bytes or less.

Receiving or transmitting more data will depend on how fast the system can notify you that activity has started and how quickly your software can respond.  You must aim to transmit/receive more data before the TX/RX FIFO is emptied of the initial 16 bytes.

I'll add the change to V67.

joan2937, 2018-03-14:
I did consider adding more core support for the I2C slave but decided against.  It is a very niche need.  It would be easy to write a C pigpio module to handle slave I2C comms using the existing interface,

closed 2018-03-20 by joan2937

kandrax, 2019-05-20:
I am using pigpio and would like to change BSC_FIFO_SIZE to 32 or 64.
Where I can find that constant "BSC_FIFO_SIZE" on raspberry Pi?
Which folder, which file?
Thank you very much.

ayeteng, 2019-05-21:
https://github.com/joan2937/pigpio/commit/424ce5af6ab831cd846a31620ff69ec6486c9268

kandrax, 2019-05-21:
Hi. Thank you for the hints.

I am still not able to locate the file that has macro BSC_FIFO_SIZE.
I am using the Python library PIGPIO.
Because I am not so experienced with Linux and Raspberry Pi, my question is where I can find the file that contains this macro. Maybe it is pigpio.h, maybe in my Python case is some other file.

In order to change I2C slave buffer size of Python lib, I need to know the path like this, for example:
...root/pi/home/blabla/pigpio.xyz
... and changing macro inside will be easy afterwards.

So, does anyone know where is this file hidden in Raspberry Pi?
Thanks.

guymcswain, 2019-05-21:
The Python PIGPIO library is a client that connects (via socket interface) to the pigpio daemon which runs the pigpio C library code.  The "macro" you refer to is part of the C library.  To change it you must rebuild the pigpio library after you modify the pigpio.h file.

Is this what you want to do?

kandrax, 2019-05-21:
Thank you.
I just want to change BSC_FIFO_SIZE in some file,
I need to know WHERE is that file?
I need an exact path. I can not find it.

guymcswain, 2019-05-21:
Look for the header file in /usr/bin/pigpio or /usr/local/bin/pigpio.  Or use the find command.

kandrax, 2019-05-22:
:-(

I don't have folder: **/usr/bin/pigpio**
I don't have folder: **/usr/local/bin/pigpio**
The following command doesn't give any results: **sudo find / pigpio.h**
The following command doesn't give any results: **sudo find / pigpio**

Any other idea/command that could help me to locate the file that has definition of macro BSC_FIFO_SIZE on my Pi 3 B+?

(I have installed pytohn3-pigpio package)

guymcswain, 2019-05-22:
My bad.  You probably have /usr/bin/pigpiod - available as part of the Raspbian distribution.  This executable has the compiled pigpio library but there are no headers.  You'll need to download the pigpio library from [here](https://github.com/joan2937/pigpio/archive/master.zip), change the pigpio.h file, re-compile and install your modified version.

kandrax, 2019-05-22:
No, I don't have that folder "/usr/bin/pigpiod"  :-/

I have already installed some PIGPIO since I am using its functions in my Python script.
I did as you said. Downloaded the ZIP file to Raspberry and extracted.
I did as the "readme" says:
**make

sudo make install**

So probably, now I have double installed PIGPIO.

I was using Slave I2C Pins of Pi earlier::
CLOCK: BCM19/Pin35/Alt3
DATA: BCM18/Pin12/Alt3

The read me mentioned some other Pins:
* All the tests make extensive use of **gpio 25 (pin 22)**.    *
* Ensure that either nothing or just a LED is connected to *
* gpio 25 before running any of the tests. 

**Question:**
Am I on the right way?
Can I somehow confirm your theory that I have pigpiod without H file and already installed?

My step are given bellow.

Regards

Dorde

`**pi@raspberrypi:~/Downloads/pigpio-master $ make**

gcc -O3 -Wall -pthread -fpic -c -o pigpio.o pigpio.c
gcc -O3 -Wall -pthread -fpic -c -o command.o command.c
gcc -shared -Wl,-soname,libpigpio.so.1 -o libpigpio.so.1 pigpio.o command.o
ln -fs libpigpio.so.1 libpigpio.so
strip --strip-unneeded libpigpio.so
size     libpigpio.so
   text   data    bss    dec    hex filename
 245697   5380 598260 849337  cf5b9 libpigpio.so
gcc -O3 -Wall -pthread -fpic -c -o pigpiod_if.o pigpiod_if.c
gcc -shared -Wl,-soname,libpigpiod_if.so.1 -o libpigpiod_if.so.1 pigpiod_if.o command.o
ln -fs libpigpiod_if.so.1 libpigpiod_if.so
strip --strip-unneeded libpigpiod_if.so
size     libpigpiod_if.so
   text   data    bss    dec    hex filename
  59683   4328  49244 113255  1ba67 libpigpiod_if.so
gcc -O3 -Wall -pthread -fpic -c -o pigpiod_if2.o pigpiod_if2.c
gcc -shared -Wl,-soname,libpigpiod_if2.so.1 -o libpigpiod_if2.so.1 pigpiod_if2.o command.o
ln -fs libpigpiod_if2.so.1 libpigpiod_if2.so
strip --strip-unneeded libpigpiod_if2.so
size     libpigpiod_if2.so
   text   data    bss    dec    hex filename
  70487   4328   1984  76799  12bff libpigpiod_if2.so
gcc -O3 -Wall -pthread   -c -o x_pigpio.o x_pigpio.c
gcc -o x_pigpio x_pigpio.o -L. -lpigpio -pthread -lrt
gcc -O3 -Wall -pthread   -c -o x_pigpiod_if.o x_pigpiod_if.c
gcc -o x_pigpiod_if x_pigpiod_if.o -L. -lpigpiod_if -pthread -lrt
gcc -O3 -Wall -pthread   -c -o x_pigpiod_if2.o x_pigpiod_if2.c
gcc -o x_pigpiod_if2 x_pigpiod_if2.o -L. -lpigpiod_if2 -pthread -lrt
gcc -O3 -Wall -pthread   -c -o pig2vcd.o pig2vcd.c
gcc -o pig2vcd pig2vcd.o
strip pig2vcd
gcc -O3 -Wall -pthread   -c -o pigpiod.o pigpiod.c
gcc -o pigpiod pigpiod.o -L. -lpigpio -pthread -lrt
strip pigpiod
gcc -O3 -Wall -pthread   -c -o pigs.o pigs.c
gcc -o pigs pigs.o command.o
strip pigs

**pi@raspberrypi:~/Downloads/pigpio-master $ sudo make install**
install -m 0755 -d                             /opt/pigpio/cgi
install -m 0755 -d                             /usr/local/include
install -m 0644 pigpio.h                       /usr/local/include
install -m 0644 pigpiod_if.h                   /usr/local/include
install -m 0644 pigpiod_if2.h                  /usr/local/include
install -m 0755 -d                             /usr/local/lib
install -m 0755 libpigpio.so.1      /usr/local/lib
install -m 0755 libpigpiod_if.so.1  /usr/local/lib
install -m 0755 libpigpiod_if2.so.1 /usr/local/lib
cd /usr/local/lib && ln -fs libpigpio.so.1      libpigpio.so
cd /usr/local/lib && ln -fs libpigpiod_if.so.1  libpigpiod_if.so
cd /usr/local/lib && ln -fs libpigpiod_if2.so.1 libpigpiod_if2.so
install -m 0755 -d                             /usr/local/bin
install -m 0755 pig2vcd                        /usr/local/bin
install -m 0755 pigpiod                        /usr/local/bin
install -m 0755 pigs                           /usr/local/bin
if which python2; then python2 setup.py install ; fi
/usr/bin/python2
running install
running build
running build_py
creating build
creating build/lib.linux-armv7l-2.7
copying pigpio.py -> build/lib.linux-armv7l-2.7
running install_lib
copying build/lib.linux-armv7l-2.7/pigpio.py -> /usr/local/lib/python2.7/dist-packages
byte-compiling /usr/local/lib/python2.7/dist-packages/pigpio.py to pigpio.pyc
running install_egg_info
Writing /usr/local/lib/python2.7/dist-packages/pigpio-1.42.egg-info
if which python3; then python3 setup.py install ; fi
/usr/bin/python3
running install
running build
running build_py
creating build/lib
copying pigpio.py -> build/lib
running install_lib
copying build/lib/pigpio.py -> /usr/local/lib/python3.5/dist-packages
byte-compiling /usr/local/lib/python3.5/dist-packages/pigpio.py to pigpio.cpython-35.pyc
running install_egg_info
Writing /usr/local/lib/python3.5/dist-packages/pigpio-1.42.egg-info
install -m 0755 -d                             /usr/local/man/man1
install -m 0644 *.1                            /usr/local/man/man1
install -m 0755 -d                             /usr/local/man/man3
install -m 0644 *.3                            /usr/local/man/man3
ldconfig
pi@raspberrypi:~/Downloads/pigpio-master $ 

**sudo ./x_pigpio**
2019-05-13 09:25:52 initInitialise: Can't lock /var/run/pigpio.pid
pigpio initialisation failed.`

kandrax, 2019-05-22:
The ZIP file already had BSC_FIFO_SIZE   512

In my python script I still use the not changed line "import pigpio",
How do  know whether I am now using the thing you told me to install?
I tested and it seems the I2C fetches only 16 bytes as before. How to deactivate the old Pigpio and switch to the new Pigpio I just installed?

Thanks

guymcswain, 2019-05-22:
Run the command `which pigpiod`  will show you the path to pigpiod.

You need to reboot to stop/restart the service to pick up the newly installed version.

Did you edit the pigpio.h with your custom setting prior to `make`?

guymcswain, 2019-05-22:
In Python script `import pigpio` just imports the client script - has nothing to do with the pigpiod you are installing.

joan2937, 2019-05-22:
This change has been in pigpio since V67.  It is pretty unlikely that the change makes any difference to a Python script.  The chances of getting your software to read the buffer in a timely fashion are remote.  See https://github.com/joan2937/pigpio/issues/160#issuecomment-372666790

Your best chance at being able to respond in a timely fashion is to make I2C transfers as slow as possible.  The I2C master decides the bit rate.