joan2937/pigpio #161: Method on pigs/socket to set many gpio at once

MaitreDede, 2017-09-21:
Hi,
Is there a method to write to many gpios at once ?
For example, a method that takes 2 parameters, one uint32 that is the ports it will write ([bits](http://abyz.co.uk/rpi/pigpio/pigs.html#bits)), and one uint32 that are the values it will write (0 or 1).

In my case, this will speed up things a lot, since I will have only one packet to send instead of 10 (or more)...

Maybe also the same for a "read" ?

For the C interface, it seams that there are the `gpio(Read/Write)_Bits_(range)_(Clear/Set)`...

guymcswain, 2017-09-21:
pigs - fifo/pipe interface
```
$pigs bs1 bits
$pigs bc1 bits
$pigs br1
bits
```
python - socket interface
```
set_bank1(bits)
clear_bank1 (bits)
bits = read_bank1()
```

MaitreDede, 2017-09-21:
Ok, I have just seen them... Need more coffee. Thanks @guymcswain :)

closed 2017-09-21 by MaitreDede

guymcswain, 2017-09-21:
We are wide awake at this time in Texas 😄.  Are you planning to use these methods to drive 10 TLC5947 devices?  I'm interested to know if having the SCLK pulses stretched on occasion, due to the OS preemption, will be a problem.

MaitreDede, 2017-09-21:
@guymcswain New Caledonia, just starting a new day...
I am not using these to drive my TLC5947, I am using the BSPIx (bitbang spi) methods :
`BSPIO <latch> <any> <data> <clock> <speed> <flag>` with :
- latch : latch pin
- data : data pin
- clock : clock pin
- speed : 250000 (max according to doc)
- flag : 0
- any : any unused pin, (we don't need to read anything).

Then `BSPIX <latch> <buffer>` And `BSPIC <latch>`
The hard part was to build buffer according to what TLC5937 wants.

MaitreDede, 2017-09-22:
@guymcswain https://github.com/MaitreDede/RaspberryPi.Libs/blob/master/RaspberryPi.PiGPIO.Drivers/Dede/TLC5947.cs